solvespace/solvespace #745: Compiling on OpenBSD fails due to unknown 'locale_t'

lenzj, 2020-10-17:
### System information

SolveSpace version: git master 0288c0a

Operating system: OpenBSD 6.7 amd64

### Expected behavior

Successful compile / build of git master on OpenBSD per instructions in the README.md

### Actual behavior

The make step is unable to complete due to errors with unknown type name 'locale_t'.
```
$ make
...
Scanning dependencies of target flatbuffers                                                                                                                                                    
[  8%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/idl_parser.cpp.o                                                                                                  
[  8%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/idl_gen_text.cpp.o                                                                                                
[  8%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/reflection.cpp.o                                                                                                  
[  9%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/util.cpp.o                                                                                                        
In file included from /home/username/packages/apps/solvespace/extlib/flatbuffers/src/util.cpp:41:                                                                                                 
In file included from /home/username/packages/apps/solvespace/extlib/flatbuffers/include/flatbuffers/util.h:25:                                                                                   
In file included from /usr/include/c++/v1/sstream:174:                                                                                                                                         
In file included from /usr/include/c++/v1/ostream:138:                                                                                                                                         
In file included from /usr/include/c++/v1/ios:216:                                                                                                                                             
In file included from /usr/include/c++/v1/__locale:32:                                                                                                                                         
In file included from /usr/include/c++/v1/support/newlib/xlocale.h:26:                                                                                                                         
/usr/include/c++/v1/support/xlocale/__strtonum_fallback.h:24:64: error: unknown type name 'locale_t'                                                                                           
                                                char **endptr, locale_t) {
...
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.                                                                           
*** Error 1 in . (extlib/flatbuffers/CMakeFiles/flatbuffers.dir/build.make:102 'extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/util.cpp.o')
*** Error 2 in . (CMakeFiles/Makefile2:325 'extlib/flatbuffers/CMakeFiles/flatbuffers.dir/all': make -f extlib/flatbuffers/CMakeFiles/flatbu...)
*** Error 2 in /home/username/packages/apps/solvespace/build (Makefile:152 'all': make -f CMakeFiles/Makefile2 all)
```
The full output from the cmake step is below:
```
$ cmake .. -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_FILE_PREFIX_MAP
-- Performing Test HAS_FILE_PREFIX_MAP - Failed
-- Using in-tree libdxfrw
-- Using in-tree flatbuffers
-- Looking for strtof_l
-- Looking for strtof_l - not found
-- Looking for strtoull_l
-- Looking for strtoull_l - not found
-- Using in-tree q3d
-- Using in-tree mimalloc
-- 
-- Library base name: mimalloc
-- Build type       : release
-- Install directory: /usr/local/lib/mimalloc-1.6
-- Build targets    : static
-- 
-- Looking for backtrace
-- Looking for backtrace - not found
-- Found Backtrace: /usr/local/lib/libexecinfo.so.2.0  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found ZLIB: /usr/lib/libz.so.5.0 (found version "1.2.3") 
-- Found PNG: /usr/local/lib/libpng.so.18.0 (found version "1.6.37") 
-- Found Freetype: /usr/X11R6/lib/libfreetype.so.30.0 (found version "2.10.1") 
-- Checking for module 'cairo'
--   Found cairo, version 1.16.0
-- Found OpenGL: /usr/X11R6/lib/libGL.so.17.1   
-- Could NOT find SPACEWARE (missing: SPACEWARE_LIBRARY SPACEWARE_INCLUDE_DIR) 
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.12.4
-- Checking for module 'json-c'
--   Found json-c, version 0.13.1
-- Checking for modules 'gtkmm-3.0>=3.18;pangomm-1.4;x11'
--   Found gtkmm-3.0, version 3.24.2
--   Found pangomm-1.4, version 2.42.1
--   Found x11, version 1.6.9
-- Configuring done
-- Generating done
-- Build files have been written to: /home/username/packages/apps/solvespace/build
```
### Additional information

In addition to README.md instructions I also installed gettext-tools using pkg_add to address following warning during cmake step:
```
$ cmake .. -DCMAKE_BUILD_TYPE=Release
...
CMake Warning at CMakeLists.txt:315 (message):
  Gettext not found, translations will not be updated
...
```

ppd, 2020-10-17:
Does a more current `flatbuffers` than f73d205 compile correctly on OpenBSD?

lenzj, 2020-10-17:
Thanks for the suggestion.

```
$ cd extlib/flatbuffers/
$ git checkout master
$ git pull
```
After doing this I started again from a clean build dir and a different compile error occurred.
```
Scanning dependencies of target flatbuffers
[  7%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/idl_parser.cpp.o
[  8%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/idl_gen_text.cpp.o
[  8%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/reflection.cpp.o
[  9%] Building CXX object extlib/flatbuffers/CMakeFiles/flatbuffers.dir/src/util.cpp.o
In file included from /home/username/packages/apps/solvespace/extlib/flatbuffers/src/util.cpp:41:
In file included from /home/username/packages/apps/solvespace/extlib/flatbuffers/include/flatbuffers/util.h:26:
In file included from /usr/include/c++/v1/sstream:174:
In file included from /usr/include/c++/v1/ostream:140:
In file included from /usr/include/c++/v1/locale:208:
/usr/include/c++/v1/__bsd_locale_fallbacks.h:123:17: error: use of undeclared identifier 'vasprintf'; did you mean 'vsprintf'?
    int __res = vasprintf(__s, __format, __va);
                ^
/usr/include/c++/v1/cstdio:125:9: note: 'vsprintf' declared here
using ::vsprintf;
        ^
...
```
I then tried checking out the different flatbuffer release tags as shown below and repeating from a clean build dir each time.
```
$ cd extlib/flatbuffers/
$ git checkout v1.12.0
```
v1.12.0 - Same locale_t error message.
v1.11.0 - Success!  The full build completed and I was able to do a quick install and run the executable.

So it looks like an older version of flatbuffers works on OpenBSD.   f73d205 appears to be a commit between v1.11.0 and closer to v1.12.0.  Thanks again for the suggestion!

ppd, 2020-10-17:
I think you should report that upstream.

lenzj, 2020-10-17:
Good point.  As I get some time this week I will try and bisect to figure out the offending commit and see if I can identify a patch to fix it.  At a minimum I will report the issue upstream.

lenzj, 2020-10-17:
I bisected this and realized the offending code is fairly simple / straightforward.  It started with commit https://github.com/google/flatbuffers/commit/625338d095cdbcd89fc98a07cd41db661826eeb3.  Defining the following in flatbuffers/src/util.cpp is what creates the problem compiling under OpenBSD.

_POSIX_C_SOURCE
_XOPEN_SOURCE

By manually commenting those two definitions out I am able to successfully compile flatbuffers on the offending commit and master.  I've reported the issue upstream (link below) and am looking into the best way to resolve this without breaking builds on other platforms.

https://github.com/google/flatbuffers/issues/6185

ppd, 2020-10-17:
Very nice. Thank you for the investigation.

phkahler, 2020-10-25:
So they changed it to force people to be more explicit: https://github.com/google/flatbuffers/pull/5529

lenzj, 2020-10-25:
It's not clear to me why the definition of _POSIX_C_SOURCE and _XOPEN_SOURCE are needed on all platforms in flatbuffers?  It solves a compile issue on a windows system compiling using gcc targeting QNX embedded OS. 
 Note that since 5529 it's since been updated in https://github.com/google/flatbuffers/commit/7571b2ac56271777e0c5fde11371c0d054e9c1f5 to define them as follows which from my understanding sets up a POSIX 2008 environment.
```cpp
#  define _POSIX_C_SOURCE 200809L
#  define _XOPEN_SOURCE 700L
```

Regardless when these are defined it won't compile on OpenBSD.  Further details and discussion can be seen in the pull request comments at https://github.com/google/flatbuffers/pull/6205

lenzj, 2020-11-12:
This issue has been resolved upstream as part of pull request google/flatbuffers#6205 and associated commit google/flatbuffers@de1f034

I will submit a solvespace pull request to use the new version of flatbuffers.

closed 2020-11-18 by ruevs