solvespace/solvespace #759: Performance Regression

phkahler, 2020-10-21:
### System information

For SovleSpace version see the bad commit listed below.
Operating system: Fedora 32

### Expected behavior

Performance should be similar to before

### Actual behavior

After commit:  https://github.com/solvespace/solvespace/commit/ab10e38d44b26de3ee64b60db0d154278d1ee8ac

The performance test I've been using went from 15 seconds to 193 with a Release + OMP build.

### Additional information

I'm going to try to make the additional improvements suggested in the fix rather than revert it.

For reference I take the attached file and change the number of repeats in the last group from 4 to 40. Wait for a bit and the Generate::DIRTY time is printed in the terminal. BTW it has always given a few bad edges.
[perf_test.zip](https://github.com/solvespace/solvespace/files/5412461/perf_test.zip)

ruevs, 2020-10-21:
Indeed. After https://github.com/solvespace/solvespace/pull/737 https://github.com/solvespace/solvespace/commit/ab10e38d44b26de3ee64b60db0d154278d1ee8ac the time to switch from "g005" to "g006" increases from 12s. to 45s. on a release build with ENABLE_LTO, on this "little model" [PrismConeNURBSNormalsTangents300.zip](https://github.com/solvespace/solvespace/files/5414683/PrismConeNURBSNormalsTangents300.zip) I've been using (don't even try it in debug mode ;-) ).

chord tolerance 0.5
max segments 40

On the other hand your model (as is) is pretty fast. I presume you increase the "repeat" times to much more than 4 in the rotate group (g007)? With 20 repeats switching from g006 to g007 takes 8s. and 40s. respectively.

phkahler, 2020-10-21:
> I presume you increase the "repeat" times to much more than 4 in the rotate group (g007)?

Yes. My test is to change it from 4 to 40.  Without the "fix" 15.5 seconds for me. With the fix 193s. Not sure why I never added OMP to SShell::CopyCurvesSplitAgainst() but that's easy and drops the version before the regression to 10 seconds.

I have not mentioned LTO, but that takes the original 15s down to 11, so combining that with OMP in that function should get it down around 7. Not bad for something that started around 2 minutes. But the regression.... I can't justify that performance penalty for a so-so NURBS fix and I really want the fix, so today I will be busy ;-)

closed 2020-10-23 by phkahler