solvespace/solvespace #770: Length dimension switches orientation

nirvdrum, 2020-10-23:
### System information

SolveSpace version: 3.0~0761339e

Operating system: Windows 10 Pro

### Expected behavior

I've attached a design file that has 0 DoF. In _g003-shell-sketch_ I have a length dimension of 25.0mm that sets the height for the overall object. In _g009-inner-shell-sketch_ I have another height dimension of 2.50mm intended to offset an inner shape from the top of the object.

If I got back to _g003-shell-sketch_ and change the length dimension from 25.0mm to 11.0mmm, the aforementioned 2.50mm dimension flips orientation. I.e., instead of positioning the inner shape 2.50mm below the top of the shell it extends the height of that inner shape such that it extends beyond the top of the shell by 2.50mm.

I would not expect changing one dimension to alter the orientation of an unrelated dimension. Moreover, I would not expect a drawing with 0 DoF to alter its orientation.

### Actual behavior

Instead of having a shape that's truly parametric, I have a case where if I reduce the size of the shape the orientation of a length dimension switches on me, which subsequently breaks the final model.

### Additional information

[Drawing](https://github.com/solvespace/solvespace/files/5427119/corsair-carbide-540-case-foot.zip)

ruevs, 2020-10-23:
Yes, sounds like a duplicate of:
https://github.com/solvespace/solvespace/issues/76
https://github.com/solvespace/solvespace/issues/440
https://github.com/solvespace/solvespace/issues/676

phkahler, 2020-10-23:
This is a duplicate. It is a tricky problem because the sketch satisfied the constraints in both cases. The problem is one of the mathematically correct solutions to the constraint equations isn't what the user wanted.

Thinking of adding #76 to the 4.0 milestone even if it's just a few constraint type like point-to-line distance in 2d. I'm doubtful we can write a point-to-point-distance constraint in a way that doesn't do this.

ruevs, 2020-10-23:
An (annoying) workaround is to remove the old constraint, drag the object close to the position you want and then create the new constraint.

closed 2020-10-23 by ruevs

ruevs, 2020-10-23:
Or in your case when the 2.5mm dimension in g009 flips "up" just "very fast drag it" down to flip it back.

nirvdrum, 2020-10-23:
I get that both options are valid. I would expect that there is 1 DoF listed if the point is allowed to flip around. I'll try fast dragging to reset it. My previous attempts at dragging it down hadn't worked, so I've resorted to removing and re-adding the constraint.