solvespace/solvespace #772: 3.0 Release prep

phkahler, 2020-10-23:
Thought I'd make a place to discuss things that need to be buttoned up prior to a beta or release candidate. I'll start. In no particular order:

Answer the question: Does Windows get 2 builds? with and without OMP.

Update github and the main site to let Windows users know they need:
https://www.microsoft.com/en-us/download/details.aspx?id=48145  (is that version sufficient?)
Because of OpenMP. Or that it's needed for one of the downloads.

Finalize the Travis changes so we can do this ;-)

Figure out what to do with PR #493  - take that discussion ot the PR I guess.

Anything missing in the changelog?

Feel free to add more in the comments below.

Obviously not committing to new features for 3.0 at this point, but new bugs deserve some consideration and triage. Old issue will still be there, but I think we're in pretty good shape - better than 2.3 is so many ways, and worse in none that I can think of.

phkahler, 2020-10-23:
Updated version strings anywhere?

ruevs, 2020-10-23:
Once the release is ready - update the web site - conservatively - but I do have a few things in mind. Once the GitHub project is updated does anyone know how to put it up on solvespace.com? Maybe @jwesthues still wants to do this himself?

jwesthues, 2020-10-23:
There's a git repo for the website in this same organization; feel free to make the changes. Let me know when you're ready to deploy, and I can pull to live.

vespakoen, 2020-10-24:
Some things that come to mind:

- Test the binaries ;)
- Make sure Windows and macOS builds have no shared libraries?
- Test the macOS build on a 'non-infected' computer, preferably with an older OS version?
- Make sure the cli is also working (we could test this in CI as well)
- Make sure macOS notarising is working (I saw a warning about a secret not being escaped, I will see what happens with the current build and if it fails I will re-upload the secrets)
- Add (an issue for) documentation for the new features
- SolveSpace as a library, is this still possible / being built? (http://solvespace.com/library.pl)
- Possibly make a windows installer (using NSIS?) that also includes the OpenMP dll?

ppd, 2020-10-24:
Before & after release:
- [ ] Decide whether we want to keep the git hash in the snap version for tagged commits
- [ ] Promote the stable, tagged revision in the Snap Store to channel `stable`
- [ ] Propose solvespace for "Featured Snaps Friday" (https://forum.snapcraft.io/t/call-for-suggestions-featured-snaps-friday-23rd-october-2020/20637)

phkahler, 2020-10-24:
@ppd Decide whether we want to keep the git hash in the snap version for tagged commits

Is there a reason not to keep that?

ppd, 2020-10-25:
@phkahler just optics, I guess. Stable could be just "3.0". On the other hand, the hash is very useful for quickly mapping a revision to a commit, so lets keep it.

phkahler, 2020-10-25:
@vespakoen  Test the macOS build on a 'non-infected' computer, preferably with an older OS version?

It doesn't work on my Air with 10.12.6 Sierra:
Dyld Error Messaage:
  Library not loaded: /usr/local/opt/llvm@9/lib/libomp.dylib
  Referenced from: /Volumes/VOLUME/SolveSpace/app/Contents/MacOS/SolveSpace
  Reason: image not found

There is no llvm@9 folder on my machine.

vespakoen, 2020-10-26:
Yep I also noticed, I am working on a fix...

vespakoen, 2020-10-26:
@phkahler can you try with the latest edge? would be interesting to see if it works on Sierra

phkahler, 2020-10-26:
> @phkahler can you try with the latest edge? would be interesting to see if it works on Sierra

It seems to work. I open the DMG and it does whatever it does and a folder opens with just SolveSpace. Running that gives me one pop-up about opening random stuff from the internet. I say open anyway and it seems to just work. I'll give a better going over later when we're a little more finalized.

vespakoen, 2020-10-26:
Yeah that sounds correct, we can improve the installation method on macOS by adding a symbolic link to the Applications folder and include a background, but the process for this is... hairy and complicated to say the least.

The warning for stuff downloaded from the internet is also expected, the only way around this is to release through the App Store, not sure if this is worth it though...

phkahler, 2020-10-29:
Is the Windows build 32 or 64bit?

jwesthues, 2020-10-29:
> Is the Windows build 32 or 64bit?

I guess if we offer a 64-bit version, then it's still customary to also offer a 32-bit version? So the lazy option would be 32 only. I'm not sure if anyone's ever benchmarked, but I doubt there'd be a big speed advantage to the 64.

ppd, 2020-10-29:
@phkahler `-DCMAKE_GENERATOR_PLATFORM="Win32"` ⇒ 32-bit

ruevs, 2020-10-29:
32 bit only in my opinion. SolveSpace will be unusably slow long before it consumes 2GB of RAM, which is the limit for 32bit Win32 processes. At this point there is absolutely no advantage to a 64bit build (but it does work).

phkahler, 2020-10-29:
I just wanted to know which version of the vc_redist stuff to mention for the OMP build ;-) x86 it is.

phkahler, 2020-11-01:
Are we good to make an RC1? Once we know what to link to I (someone) can add a table entry on the download page and have JW push the changes there.

ruevs, 2020-11-01:
To me master  looks the best so far (which admittedly it is most of the time), all the milestone issues are done. I think an RC1 linked from the official page is in order. Many more people need to run it to find more subtle bugs. So I am all for it.

So what is the procedure? Make a "3.0_RC" branch, change the version there and tag it? Then wait for Travis to build it?

phkahler, 2020-11-01:
I will defer to @ppd but I don't think a branch prior to 3.0 is really warranted. A branch at 3.0 might be nice in case we want to back port important fixes once other more invasive changes start happening. BTW @ppd if you're in agreement feel free to pull the trigger.

jwesthues, 2020-11-02:
Yeah--I'd probably make a branch at 3.0 in case we need to backport, and tag the release candidate(s). We should test the RC among ourselves, then post it on the download page (clearly labeled as experimental/beta/whatever), and then if all seems well after a couple months relabel it as the latest official and recommended release.

ppd, 2020-11-02:
My personal preference: 
1. tag master HEAD `v3.0.rc1` right now
2. polish the release on master
3. tag master HEAD `v3.0`
4. branch there to `3.x`

So a yes to the workflow of @phkahler & @jwesthues.

ruevs, 2020-11-02:
Since we do not have anything "urgent" with significant risky changes to do on "master" @ppd 's suggestion to "stabilize" (do the RCs) on master should be fine. It will save some potential merging back-and forth. But we will have to avoid doing risky/drastic/major things on master until we release 3.0 and branch 3.x

phkahler, 2020-11-18:
@ppd  time for an RC. Can you kick that off and see that it arrives on the download page OK?  Then we'll have JW update the web site.

ppd, 2020-11-18:
Alright. It needs a little Travis fix, but then we should be good to go.

phkahler, 2020-11-18:
Yeah it put source code but not executables ;-)

@jwesthues  Can you make the recent website changes live? Oh, and I guess we need to add another table entry for the 3.0-RC1 release - I'm not able to do that for another 10 hours so that would be good too.

ppd, 2020-11-18:
Okay. Seems like that worked except for that weird snap glitch.  I think it has to do with the fact that it's a tagged commit. Most likely a braindead error in `snapcraft.yaml`. I'll investigate.

jwesthues, 2020-11-19:
I've added the table entry, and pulled to live. Thanks, and let me know if anyone sees any issues.

ruevs, 2020-11-19:
In the "Platforms" column - (unfortunately) it does not work on XP any more. So it should be  "Windows Vista–10, Linux, Mac"

It does still work on XP if one compiles the OpenGL 1 front end version and goes through some extra hoops with the SDK and compiler. Probably I'll make an unofficial XP release on my fork when 3.0 is final.

0x0203, 2020-11-19:
Not sure if you want this here or in a separate issue, but building (for Linux) from the packaged solvespace-3.0.rc1.tar.gz on the downloads page doesn't work since it doesn't include the required submodules. I don't know if it's supposed to, but cmake won't complete without q3d, libdxfrw, mimalloc, and flatbuffers. And since this isn't a git clone, running the 'git submodule update --init ...' command won't work.

Incidentally, the build instructions for Linux only mention the git submodule update step for libdxfrw. Including mimalloc, flatbuffers and q3d if they really are required might be useful.

jwesthues, 2020-11-19:
@ruevs Thanks, good catch, merged and should be live.

Symbian9, 2020-11-22:
### Before 3.0 release:

- [ ] Revert https://github.com/solvespace/solvespace/commit/b2418a0324fa4e4ed498a46aa0946b67a267ebf0
- [ ] Request removing of https://cla-assistant.io/solvespace/solvespace

ruevs, 2020-11-22:
> ### Before 3.0 release:
>     * [ ]  Revert [b2418a0](https://github.com/solvespace/solvespace/commit/b2418a0324fa4e4ed498a46aa0946b67a267ebf0)
>     * [ ]  Request removing of https://cla-assistant.io/solvespace/solvespace

@Symbian9 3.0 does not depend on these at all.

phkahler, 2020-12-12:
Should we have an RC2?

I'd really like to get 3.0 Final, but if we want an RC2 I'd probably merge the rebased Eigen stuff, the Apple M1 if it gets there, look into the SpaceMouse issue, and one last minute feature I'd like, and I think @ruevs tagged something for 3.0.

Symbian9, 2020-12-12:
> Should we have an RC2?

Yes! Especially to give more time for translator to update/add new locales — `3.0rc2` should be *"Call for Translators!"*.

Symbian9, 2020-12-12:
> **Before 3.0 release:**
> 
> * [ ]  Revert [b2418a0](https://github.com/solvespace/solvespace/commit/b2418a0324fa4e4ed498a46aa0946b67a267ebf0)
> * [ ]  Request removing of https://cla-assistant.io/solvespace/solvespace

> 3.0 does not depend on these at all.

Removing CLA signing requirement for contributing *before* `3.0` is really important!

@ruevs, please, read comment by @jwesthues:

 - https://github.com/solvespace/solvespace/issues/816#issuecomment-735278731

rpavlik, 2021-01-15:
So the Debian freeze for Bullseye is rapidly approaching (first freeze is already past) - a Debian Science maintainer is updating the package, but since rc1 was the last tag, that's what's getting packaged. It would be great to get an RC2 soon if nothing else but for getting into Debian, since it's upstream of so much.

Looks like there's currently a build failure on armel due to libdxfrw? https://buildd.debian.org/status/fetch.php?pkg=solvespace&arch=armel&ver=3.0.rc1%2Brepack1-4&stamp=1610723096&raw=0 The overall status page is here: https://tracker.debian.org/pkg/solvespace

phkahler, 2021-01-15:
The only thing I need prior to RC2 is to update the .pot file for translations. I can maybe do that tonight or this weekend, or anyone else can do it and make a PR which I can merge from phone. I want that step done so translators only need to update their particular locale after the RC.

In an ideal world we'd figure out the one remaining GTK oddness with a 3.0 label, and the SpaceMouse on Windows. But the GTK one is actually (I think) an old problem so we can leave it. SpaceMouse would be a shame to lose even though it seems uncommon with FLOSS tool users.

After RC2 we're looking at just translations and maybe some icon updates.

Symbian9, 2021-01-15:
> The only thing I need prior to RC2 is to update the .pot file for translations. I can maybe do that tonight or this weekend, or anyone else can do it and make a PR which I can merge from phone.

I'm ready for updating Ukrainian locale!

rpavlik, 2021-01-15:
I tried to do the translations update, but I'm new at it so if I goofed it up  just let me know or close it.

ruevs, 2021-01-16:
> So the Debian freeze for Bullseye is rapidly approaching (first freeze is already past) - a Debian Science maintainer is updating the package, but since rc1 was the last tag, that's what's getting 

Definitely worth making a rc2 for this!

It's impressive (but not surprising) how much "infrastructure" is involved in Debian packages!
https://salsa.debian.org/science-team/solvespace/-/commits/master/

From the above it seems there may be a mimalloc cmake change that we may want to take? To use the CXX instead of the C compiler?

> Looks like there's currently a build failure on armel due to libdxfrw? https://buildd.debian.org/status/fetch.php?pkg=solvespace&arch=armel&ver=3.0.rc1%2Brepack1-4&stamp=1610723096&raw=0 The overall status page is here: https://tracker.debian.org/pkg/solvespace

I can not understand what failed... I'll look again later on a computer...

ruevs, 2021-01-16:
Making rc2 will also reduce the Debian package by one patch
https://salsa.debian.org/science-team/solvespace/-/blob/master/debian/patches/20_use_system_flatbuffers.patch
Since we removed flatbuffers.

rpavlik, 2021-01-16:
Yes, I saw that patch when I reviewed the state of the package, and thought the same thing. That will simplify things indeed.

Once the package passes all the Debian arches, we can upstream the patches that are appropriate, if any. I think the CXX mimalloc  change was reverted but not 100% sure: I had pinged that dev to ask if he'd have time to look at this package, and if not, I'd do it. Didn't get a reply, but saw he started in on it, so I left him to it rather than trying to race.

phkahler, 2021-01-16:
I believe we are ready for RC2.  @ppd or @rpavlik ?

rpavlik, 2021-01-16:
Seems ok to me. I filled an issue with the mailing list email I got today about it, seems there is a build issue with mimalloc on arm

ppd, 2021-01-17:
> I believe we are ready for RC2. @ppd

Yes, go for it.

ruevs, 2021-01-17:
@phkahler only consider the suggestion here:
https://github.com/solvespace/solvespace/issues/890#issuecomment-761824614
Do you guys think that "Fade Previous Solid Groups" is better?

Symbian9, 2021-01-17:
> Do you guys think that "Fade Previous Solid Groups" is better?

Yes!

But also built-in style `s00d-#def-dim-solid` should be renamed to `s00d-#def-fade-solid`

rurban, 2021-01-17:
No opinion on def or not. Both sound fine to me. The shorter might be a bit more appealing.

phkahler, 2021-01-17:
@ppd I tried the "Draft a new Release" button, filled in a name and a couple words, checked the checkbox (this is a pre-release). And nothing happened. What's the correct procedure?

ruevs, 2021-01-17:
Give me a few more minutes... I'm preparing the non anti-aliased icons...

ruevs, 2021-01-17:
@phkahler https://github.com/solvespace/solvespace/pull/897

ppd, 2021-01-17:
@phkahler That's the correct procedure. Did you enter a tag name?

phkahler, 2021-01-17:
> @phkahler That's the correct procedure. Did you enter a tag name?

I did. You want to give it a go?

phkahler, 2021-01-18:
@jwesthues   I updated solvespace-web to point at the RC2. Can you make that live?

Now it's just translations and icons ;-)

jwesthues, 2021-01-18:
Excellent, just pulled to live.

phkahler, 2021-01-18:
Oops. I didnt update the text, just the table and link.

jwesthues, 2021-01-18:
No worries, let me know if you need anything else.

rpavlik, 2021-01-20:
I have made debian package changes to include rc2 and the patch for mimalloc, as well as for being able to use system three.js (needed some changes to go from  r76 to r111, will upstream those eventually). Have reached out to the person who did the maintainance of the package last week requesting review/sponsorship. It seems to work when I use it...

Symbian9, 2021-01-20:
>  I have made debian package changes... Have reached out **to the person who did the maintainance of the package** last week requesting review/sponsorship

Are you mean @AlexanderP?

- https://notesalexp.org/packages/en/buster/amd64/solvespace/

[43 more replies]

closed 2021-04-22 by phkahler