solvespace/solvespace #773: UBSan on macOS finds problems in Cairo

ruevs, 2020-10-23:
```
/Users/travis/build/solvespace/solvespace/extlib/cairo/src/cairo-image-compositor.c:133:34: runtime error: left shift of 255 by 24 places cannot be represented in type 'int'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/travis/build/solvespace/solvespace/extlib/cairo/src/cairo-image-compositor.c:133:34 in 
/Users/travis/build/solvespace/solvespace/extlib/cairo/src/cairo-fixed-private.h:64:14: runtime error: left shift of negative value -8388608
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/travis/build/solvespace/solvespace/extlib/cairo/src/cairo-fixed-private.h:64:14 in 
/Users/travis/build/solvespace/solvespace/extlib/cairo/src/cairo-fixed-private.h:64:14: runtime error: left shift of negative value -31
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/travis/build/solvespace/solvespace/extlib/cairo/src/cairo-fixed-private.h:64:14 in 
```

Because of https://github.com/solvespace/solvespace/pull/771 sanitizers will not run Cairo (and other externals) any more and this will get forgotten so this bug is to remind us.
This was discovered here: https://github.com/solvespace/solvespace/pull/766#issuecomment-714612543

When this bug is fixed I think https://github.com/solvespace/solvespace/commit/68b1abf77f5c8858659a51b8bdbf760effb53b66 should be reverted.

ruevs, 2020-10-23:
```
/usr/local/opt/llvm@9/bin/../include/c++/v1/vector:904:13: runtime error: load of value 238, which is not a valid value for type 'bool'

SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/local/opt/llvm@9/bin/../include/c++/v1/vector:904:13 in 

/bin/sh: line 1:  7869 Abort trap: 6           ../bin/solvespace-testsuite
```

ppd, 2020-10-23:
Hm. Do you think it would be wiser to revert it right now and rather work with suppression declarations (https://clang.llvm.org/docs/UndefinedBehaviorSanitizer.html#issue-suppression)?

ppd, 2020-10-24:
With @vespakoen's changes in #766, 68b1abf serves no purpose any more. Therefore, I propose to revert it right now (#775) and to investigate suppression declarations at a later time.

ruevs, 2020-10-24:
Agreed.

closed 2020-10-26 by phkahler