solvespace/solvespace #774: Github token in CI seems to be invalid

vespakoen, 2020-10-24:
@jwesthues can you create a new one for us?

vespakoen, 2020-10-24:
Just realised that this probably happened because of the move from travis-ci.**org** to travis-ci.**com**

jwesthues, 2020-10-24:
So do you still need the new token? Let us know if yes, and either @phkahler or I can take a look.

ppd, 2020-10-24:
I don't think the migration to travis-ci.com is the reason. All secrets are supposed to be migrated, and e.g. the snap secrets work just fine.

phkahler, 2020-10-24:
Can it be related to quoting vs not quoting? I pasted the new secrets for the macOS stuff without quotes which IIRC you guys had to account for. Did the others get unquoted at that time?

vespakoen, 2020-10-24:
This is actually a bit of a different issue as the regular build secrets.

The github deploy key is encryped using the travis cli and is put directly into the travis.yml config.

When encrypting the key, there is an option to choose for --org or --com, and basically, a ket generated for travis .org wont work on .com.

So, we need a new github key as described here: https://docs.travis-ci.com/user/deployment/releases/#authenticating-with-an-oauth-token

Which can be encryped as described over here: https://docs.travis-ci.com/user/encryption-keys/#usage

Hope we can sort this out :)

vespakoen, 2020-10-25:
Since we are using V2 of the deployments module, it looks like we can also add the github token as a build secret.

https://docs.travis-ci.com/user/deployment-v2/providers/releases/#environment-variables

So basically, the steps are:

- Go over here to create a new token: https://github.com/settings/tokens/new
- Make sure to give it the "public_repo" permission
- Go over here to add a secret to travis: https://travis-ci.com/github/solvespace/solvespace/settings
- Name it `GITHUB_TOKEN`

That should get our edge builds going.

ppd, 2020-10-25:
@vespakoen From your link:

> For security, it’s ideal for token to have write access limited to only repositories where Travis deploys to GitHub releases. The suggested workaround is to create a machine user — a GitHub account that is granted write access on a per repository basis.

So I think there should be machine user (https://developer.github.com/v3/guides/managing-deploy-keys/#machine-users) who has access to this one repository only.

vespakoen, 2020-10-25:
Ah yes, even better!

phkahler, 2020-10-25:
Added the GITHUB_TOKEN.

vespakoen, 2020-10-25:
Build incoming!! thanks for the hard work guys!! I am super excited about this!

https://github.com/solvespace/solvespace/releases/tag/edge

closed 2020-10-25 by vespakoen

ppd, 2020-10-25:
@phkahler Just so there's no misunderstanding. Now you've entered an OAUTH token that authorizes access to all public repos that you have access to. It's effectively you for the scopes you've authorized. It's probably still a better idea to create a user "SolveSpace Butler", who has access to `solvespace/solvespace` only.

jwesthues, 2020-10-25:
Yeah, agreed. @phkahler Feel free to create such a user and grant it privileges.

vespakoen, 2020-10-25:
Also, the edge tag triggers another build, causing an infinite loop. We should disable that :)