solvespace/solvespace #776: Snap Selection to Grid

phkahler, 2020-10-24:
### System information

SolveSpace version: 3.0~70350715

Operating system: Fedora 32

### Expected behavior

Turn on the grid. Draw some line segments. Select a point. Constrain to grid.
The point should move to either a line or vertex on the grid (I don't know which was intended).

### Actual behavior

Nothing happens. The point remains off the grid.

### Additional information

phkahler, 2020-10-26:
To clarify a bit: 
With a single line segment the endpoints can be snapped to grid, but they are not constrained and can still be dragged. Is that expected?
With multiple segments, selecting one of the non-endpoints and snapping to grid does nothing.

Going back to an old version (old enough to not show its version number) the snapped points are still movable. The non-endpoints can also be snapped, so that part seems to be the only regression.

vespakoen, 2020-10-27:
When you also select a segment next to the point it also seems to work.
When selecting just a point (in the middle of a line) I can also see it snapping but then returning back to its original position (in a split second).

phkahler, 2020-10-27:
When there are 2 segments meeting at a point there are actually 2 points there with a coincident constraint added. It's probably moving one of them to the nearest grid point but not the other, and then the solver enforces the constraint and puts it back?  Wonder when that got broken.

ruevs, 2020-10-27:
Too late to do one more bisect today... tomorrow maybe :-)

ruevs, 2020-10-28:
The regression was introduced here:

25631d4fb214239a1b46f8cd6248a3796dd2a5d4 is the first bad commit
Author: @Evil-Spirit 
Date:   Thu Jul 21 23:58:18 2016 +0600

    Choose entities to select in a way appropriate for the operation.

closed 2020-10-31 by phkahler

ruevs, 2020-11-01:
I can confirm https://github.com/solvespace/solvespace/pull/787 fixes this for me.