solvespace/solvespace #779: Font List not fully scrollable in Sketch Curves from Text

masonhensley, 2020-10-25:
Hello! Very awesome project, I'm in the early stages of replicating some classic car parts via laser cutting. I was able to design the parts with solvespace. This wasn't a biggie for me, but thought that it might be helpful to others who end up a little confused regarding font selection for Curves from Text. 

### System information
SolveSpace version: 3.0 (3.0~34328c97)

Operating system: OSX

### Expected behavior

When selecting a font text, I should be able to alphabetically scroll through the entire system font list.

### Actual behavior
Font list stops scrolling in the M's, which prevents accessing "stencil" fonts towards the end of the alphabetized list

### Additional information

[image]

phkahler, 2020-10-25:
Can not reproduce on Linux so giving this the macOS lablel.

ruevs, 2020-10-25:
Works on Windows as well.

vespakoen, 2020-10-26:
I am on macOS `10.15.5` and it is working fine over here as well.

[image]

- Can you check if it is also not working with the latest edge release from here:
https://github.com/solvespace/solvespace/releases/download/edge/SolveSpace.dmg

- What is you macOS version?

phkahler, 2020-10-26:
I can not reproduce this from an edge build:  3.0~3694c9b3   on macOS Sierra 10.12.6

@masonhensley   do you have an unusually large number of fonts on your system? All I can think of is that there's some artificial limit on how many are shown in SolveSpace?

vespakoen, 2020-10-26:
Or maybe on a non-retina display? Looks like this could be related:

[image]

https://github.com/solvespace/solvespace/commit/f324477

phkahler, 2020-10-26:
@vespakoen  you may be onto something.  Mine does not have a retina display and does not have the problem.

phkahler, 2020-10-26:
@masonhensley does it change in any noticeable way if you resize the text window?  I guess if you're checking, does it always fail to scroll to the bottom, or just on the font selection screen? click on "configuration" from the home screen for a fairly long text window to test.

vespakoen, 2020-10-26:
> @vespakoen you may be onto something. Mine does not have a retina display and does not have the problem.

I actually do have a retina display so that's probably not it...

masonhensley, 2020-11-03:
Apologies for the delayed response.

Other info that might be helpful:
- 1394 fonts installed on my system, can't figure out a good way to calculate how far into the list it's going. I basically have the system defaults and all the google fonts cached locally. (bold, italics, etc versions of a font each count independently towards the font count)
- running 4k screens on a mac mini (one horizontal, one vertical) 
- scaling the display has no impact on the list length visible (first screenshot)

- @phkahler - no, even if I try to stretch out the control panel on a vertically oriented 4k monitor, it still maxes out (sorry for obnoxiously long screenshot below)

[image]

[image]

phkahler, 2020-11-04:
@masonhensley  Thank you for the response. Knowing you have a LOT of fonts, I have been able to reproduce this on my system by having it add each font to the list 10 times (and not remove duplicates). With that change mine scrolls to somewhere in the "n" range and stops.

phkahler, 2020-11-04:
And there is the problem:
```
class TextWindow {
public:
    enum {
        MAX_COLS = 100,
        MIN_COLS = 45,
        MAX_ROWS = 2000
    };
```

Raising it to 5000 rows will fix the current problem but it will come back if someone has even more fonts. I'm not sure what's reasonable to do here. @ruevs? @ppd?

The text window seems to use several bytes per character. Making it 10000 rows would only take about 1Meg but that just seems like a hack.

ruevs, 2020-11-04:
Yes it uses

```C++
uint32_t text[MAX_ROWS][MAX_COLS];

 struct {
        char            fg;
        char            bg;
        RgbaColor       bgRgb;
        int             link;
        uint32_t        data;
        LinkFunction   *f;
        LinkFunction   *h;
}       meta[MAX_ROWS][MAX_COLS];

int top[MAX_ROWS]; // in half-line units, or -1 for unused
```

I like this hard limit and would say we should just do `MAX_ROWS = 5000`.
A person with 4900+ fonts will complain again in a few years if ever ;-) For all other text windows I think our other performance bottlenecks will kill us long before the 5000 line limit (e.g. groups, constraints etc.).

ppd, 2020-11-04:
So the user's expectation would always be to display all fonts. I just looked and I have 3100 fonts installed with just a careless extra `fonts-noto` install and some extras pulled in here and there by other packages. I've never gone "font hunting". So a pragmatic choice would be to increase to 5000 and print a little warning in case there are too many fonts.

ruevs, 2020-11-04:
And I have 164 :-D

closed 2020-11-07 by phkahler