solvespace/solvespace #782: Property Browser scrollbar returns to top

kk6mrp, 2020-10-26:
### System information

SolveSpace version: 30.~3694c9b3

Operating system: Windows 10

### Expected behavior

Scroll bar should stay in the position it was moved to.

### Actual behavior

Scroll bar and window returns to top of screen.

### Additional information

[image]

ruevs, 2020-10-27:
It stopped working here:

f324477dd06c15022670a4486bf83be3abbf48b5 is the first bad commit
commit f324477dd06c15022670a4486bf83be3abbf48b5
Author: whitequark
Date:   Thu Jul 12 19:29:44 2018 +0000

    Implement a platform abstraction for windows.

ruevs, 2020-10-27:
@phkahler @ppd @vespakoen this is fixed here https://github.com/solvespace/solvespace/pull/783 please test on Linux and macOS since this should affect all platforms.

phkahler, 2020-10-27:
@ruevs  I understand the need to test the fix on all platforms, but why did you tag the issue with the non-windows platforms?

ruevs, 2020-10-28:
> @ruevs I understand the need to test the fix on all platforms, but why did you tag the issue with the non-windows platforms?

From reading the code it appears to me that the problem should have affected all platforms. If moving the scrollbar with the mouse works on Linux and macOS before https://github.com/solvespace/solvespace/pull/783 then for sure remove the tags.

closed 2020-10-28 by phkahler

kk6mrp, 2020-10-28:
This does fix the issue for me