solvespace/solvespace #790: edge release mixes new assets & old source code, commit, date

ppd, 2020-11-02:
Currently, the edge release's notes get updated to the current master commit and its assets get updated by Travis. So far, so good.
However, the release's date and associated commit remain in their original state. The same applies to the source code assets (zip, tar.gz), which remain at the original commit.

That's not just less than ideal, but even confusing and plain incorrect. I'd prefer to re-create the GitHub release on each update of the edge tag. Ping @vespakoen.

ppd, 2020-11-02:
@vespakoen Something like

```shell
old_release=$(curl \
  -H "Accept: application/vnd.github.v3+json" \
  https://api.github.com/repos/solvespace/solvespace/releases/tags/edge \
  | jq -r ".url")

curl \
  -X DELETE \
  -H "Accept: application/vnd.github.v3+json" \
  -H "Authorization: token $GITHUB_TOKEN" \
  "$old_release"
```

in `before_deploy`. What do you think?

vespakoen, 2020-11-02:
Yes that would be great indeed, I can try adding that command to my fork so I can test it out, I'll make a PR

ppd, 2020-11-02:
Or better add it to a "prepare-edge" stage with one job only and run it before deploy. We don't want to re-create edge multiple times per commit.

ppd, 2020-11-02:
Also, tag `edge` is still pointing at 5fbb1b8f53322f4a29d86839fe88d3fd2a1fe897. The intention was to rewrite it to the latest commit, which does not seem to work either.

PR #791 (2020-11-07): Travis: add clean edge script
This fixes #790 

Thanks @ppd for providing the code!
closed 2020-11-07 by vespakoen