solvespace/solvespace #793: Suggestion: Parts Library

phkahler, 2020-11-07:
## Should we provide basic standard part sketches?
If so, should we just make some folders and zip them up? or a separate repo?

For example I've been making a few things in this .zip file including:
2020 and 4040 extrusion templates - add to assembly and extrude to length.
Raspberry Pi template - PCB in IDF format (.emn) sans parts.
M3 - M20 various standard bolts.

[SS_parts_bin.zip](https://github.com/solvespace/solvespace/files/5505412/SS_parts_bin.zip)

The bolts will want to be redone later once we can specify pitch for a helix. It may also be good to make them using a boolean with the helix done the other way. These are close to spec but not perfect.

I'm hoping to add a couple more PCBs and maybe some screws (different heads).

ruevs, 2020-11-07:
https://github.com/solvespace/solvespace/issues/330
;-)

phkahler, 2020-11-08:
@ruevs  Ugh, and I commented on that one! I'll close this.

I've also given a lot of thought to that hole-tool idea.

closed 2020-11-08 by phkahler