solvespace/solvespace #795: RFC: Should we remove Q3D export?

ruevs, 2020-11-12:
Should we remove Q3D export and with it rip out flatbuffers (which is a PITA as far as compiling in "strange" environments goes)?

https://github.com/solvespace/solvespace/pull/794#issuecomment-726353146

It seems that [Q3D](https://github.com/q3k/q3d/) export [was added](https://github.com/solvespace/solvespace/pull/384) as a way to get colorful 3D meshes into Horizon EDA https://github.com/horizon-eda/horizon/pull/236. However that pull request was not merged.

In the meantime SolveSpace now exports Step files with color that Horison displays correctly :-)
https://github.com/solvespace/solvespace/pull/767
https://github.com/solvespace/solvespace/pull/763
https://github.com/solvespace/solvespace/issues/452

So it appears to me that SolveSpace is the only program (in this universe) that exports [Q3D](https://github.com/q3k/q3d/) files and nothing imports them. On the other hand I do like this otherwise unused format - aesthetically...

 @q3k, @carrotindustries, @jwesthues, @whitequark (sorry for pinging you) what do you think?

From [the PR that added Q3D](https://github.com/solvespace/solvespace/pull/384) I know that Whitequark had plans to use flatbuffers as a basis for a future "new .slvs format" but I do not know any details - and they can always be added back if/when this happens.

ruevs, 2020-11-12:
https://freenode.irclog.whitequark.org/solvespace/2019-02-24

phkahler, 2020-11-13:
**Should we remove Q3DO export?**
Yes we should.  Q3D is apparently of no use to anyone and was invented to solve a problem that no longer exists.

ppd, 2020-11-13:
Code removed is code debugged ;-)

ruevs, 2020-11-13:
I'll prepare the change. Let it sit for a few days in case someone complains (I'll ask in the forum too) - if no one does I'll make a pull request.

ruevs, 2020-11-13:
The removal https://github.com/ruevs/solvespace/commit/491108ff3ecd6ee9ce7b82684dd7caa723fbabd6
A post in the forum https://solvespace.com/forum.pl?action=viewthread&parent=3256

q3k, 2020-11-13:
> The removal ruevs@491108f

Go ahead. If maintaining this is too much effort, absolutely do rip it out.

I’m likely the only user of this, and I don’t mind maintaining my own best effort fork instead.

phkahler, 2020-11-14:
> Go ahead. If maintaining this is too much effort, absolutely do rip it out. I’m likely the only user of this, and I don’t mind maintaining my own best effort fork instead.

OK.  If you'd like to implement STL reading in some other program I've got working import for SolveSpace here: https://github.com/phkahler/solvespace/commit/06bcfb6f32f6d8bcdc5f12da3b747edce2f37403

It can read the colors as exported by SolveSpace and some other software.

ruevs, 2020-11-18:
@q3k how are you using Q3D? Is there any software that uses the format (you created) apart from SolveSpace?

q3k, 2020-11-18:
> @q3k how are you using Q3D? Is there any software that uses the format (you created) apart from SolveSpace?

I use it in pipelines in personal software and public software forks (eg. within a forked horizon-eda, for instance, as the Q3D implementaiton there was rejected). There is no publicly available software that supports this format (yet), as I haven't yet pushed hard to make it more popular and support a few extra usecases (indexed vertex lists, PBR materials, UV coordinates). That's on my personal backlog, but to do within years :).

ruevs, 2020-11-18:
Hmmm OK. I like the (simplicity of) the format, but in a public release it serves no purpose after colors are implemented in STEP (I realize it is a chicken and egg situation). On the forum no one responded. I'll create the pull request to remove it.

If it becomes more widespread in the future, or if we need flatbuffers for something else, we can easily mainline it again.

phkahler, 2020-11-21:
Removed in #803

closed 2020-11-21 by phkahler