solvespace/solvespace #797: Regression in property browser window position memorization

pwnage101, 2020-11-15:
I hesitate to call this a regression, but I do think the old behavior was better.

### System information

SolveSpace version: 3.0~ 6c2b967790b4554021b5f0fa8f16236456f95cbb

Operating system: Debian Stable (Buster)

### Old behavior

Here's the old behavior from Solvespace stable release (v2.3):

1. With the property browser open, drag it to to the corner of the screen.
2. Press tab to close it.
3. Press tab again to open it, and the window should appear where it was left (still in the corner of the screen)

### New behavior

3. After re-opening the property browser, it always appears back in the center of the screen.

### Additional information

This is quite a minor issue, but I did feel more productive with the old behavior (with the position of the propery browser memorized).  I have not attempted to bisect to find out which commit introduced the new behavior.

phkahler, 2020-11-17:
@pwnage101  When I read this my first thought was "Wayland" and I've confirmed (Fedora 32 with Gnome) that in an Xsession the behaviour is the way you expected - text window reappears where it was. Then I switched back to Wayland and for me the text window always appears in the upper left corner (unless I put something else there - gnome tries not to overlap windows). Let us know if your problem is due to running wayland vs X.

My thoughts on this: The Wayland security model makes sense to me - an application should not have access to the rest of the desktop environment and so should not even know where it is within that environment. That means an application can not position itself. I get it. But that means it becomes the compositors job to remember where windows last were and put them there when they appear again. To be fair, having that fall on every application was a stupid design decision in the first place but I can imagine the history...

Another issue, I'm not sure if SolveSpace destroys the text window or just hides it. If it's hidden then it seems really absurd that they move it when it's made visible again. But this is way outside my familiarity with the code so perhaps someone else can comment.

To recap: **Is this actually a Wayland issue?**

pwnage101, 2020-11-17:
I'm 99% sure I'm running X.  I start my desktop environment using `startx` from tty1 and when I do `ps aux | grep -i xorg` there's clearly Xorg processes running.  Maybe this is useful information: I run a tiling window manager ([i3](https://github.com/i3/i3)).

phkahler, 2020-11-17:
@pwnage101  Then it's probably not a Wayland issue. But it behaves correctly for me with X which means I'm not reproducing the problem here.

phkahler, 2021-02-05:
A few days after upgrading to Fedora 33, I applied a patch for PR #922 which seems totally unrelated. After testing that, my terminal window contained this new message:
```
Gdk-Message: 17:45:14.723: Window 0x1370d90 is a temporary window without parent, application
will not be able to position it on screen.

(solvespace:99349): Gdk-CRITICAL **: 17:45:14.728: gdk_wayland_window_handle_configure_popup:
assertion 'impl->transient_for' failed
```

I've never seen that so want to investigate more as it seems to relate to this. It seems related to a dialog though and not the main window placement. It does suggest that placement is possible depending on window type.

robnee, 2021-02-07:
> (solvespace:99349): Gdk-CRITICAL **: 17:45:14.728: gdk_wayland_window_handle_configure_popup:
assertion 'impl->transient_for' failed

gtk_menu_popup and subsequently GtkMenu::popup are deprecated and cause [problems for Wayland](https://github.com/blueman-project/blueman/issues/1204)  There's a fix in #929