solvespace/solvespace #805: Failure to launch or draw (Linux)

0x0203, 2020-11-19:
### System information

SolveSpace version: Built from HEAD of git repo.  (a7a8ebba6)

Operating system: Linux kernel: 4.19.0-12 - Distro: MX Linux 19.3

### Expected behavior

Program runs

### Actual behavior

The program starts and two windows are created, but nothing is ever drawn in the windows. Sometimes the program dies immediately, but more often it just hangs. The below shows the console output for when it dies. When it hangs, it doesn't get to the glibmm-ERROR.

### Additional information

`json_object_from_file: error opening file /home/user/.config/solvespace/settings.json: No such file or directory`
`Missing (absent) translation for title'(new sketch)'`
`Missing (absent) translation for title'Property Browser'`

`(solvespace:5212): Gtk-WARNING **: 11:02:04.391: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown`

`(solvespace:5250): glibmm-ERROR **: 11:03:15.594: `
`unhandled exception (type std::exception) in signal handler:`
`what: std::bad_array_new_length`

`Trace/breakpoint trap`

phkahler, 2020-11-19:
What version of GTK/gtkmm are you using?  I'm not sure the minimum required, but that might be the issue.

0x0203, 2020-11-24:
Apologies for the delay; For GTK I have both 3.24.5 and 2.24.32 installed. Similarly with libgtkmm, I've got the both 3.0 and 2.4 versions installed (3.24.0 and 2.24.5).  According to the CMakeCache it's using version 3 to build.

0x0203, 2020-12-15:
Some further information: I tried running one of the debian nightly builds and got some additional error messages to print out: 
`File ./src/render/gl3shader.cpp, line 104, function CompileShader:  `
`Assertion failed: shader != 0.  `
`Message: glCreateShader failed.  `
`Aborted`

I'm running with the latest nvidia drivers for my system, everything is up-to-date and I don't seem to have a problem with other open gl apps.

0x0203, 2020-12-18:
By uninstalling the nvidia drivers, I was able to get solvespace to run. Of course running w/o the nvidia drivers prevents using dual monitor and other 3d intensive programs, but it seems to be a problem with some interaction between opengl and nvidia on my hardware and not an issue with solvespace.

closed 2020-12-18 by 0x0203