solvespace/solvespace #807: New Travis pricing plans and their relevance for SolveSpace

ppd, 2020-11-20:
I've just received an email informing me about the "Free Plan" with "10000 credits left" that I now have on Travis.

Apparently, on 2020-11-02, a new billing/pricing scheme was introduced: https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing
So rather soon after we made the switch to travis-ci.com.

There's talk about "The upcoming pricing change will affect those of you who are: Building on the macOS environment" and users being "gradually upgraded to the new pricing scheme".

So I'm not really sure what this all means for an organization that has only OSS repos to build, but surely that information is to be found somewhere; I'll have to do a little research.

ppd, 2020-11-20:
So SolveSpace as an organization has no (visible) plan set as of now: https://travis-ci.com/organizations/solvespace/plan
This means we're likely to get changed to the "Free" plan, which has 10,000 credits only. As for extra OSS credits:

>  We will be offering an allotment of OSS minutes that will be reviewed and allocated on a case by case basis. Should you want to apply for these credits please open a request with Travis CI support stating that you’d like to be considered for the OSS allotment. Please include:
> 
>  Your account name and VCS provider How many credits (build minutes) you’d like to request (should your run out of credits again you can repeat the process to request more or discuss a renewable amount)

It seems our Travis time investment would have better been spent on migrating to something else.

ruevs, 2020-11-20:
Hmm what does 10000 credits mean? 10000 minutes? A month? Which minutes? From the [last build](https://travis-ci.com/github/solvespace/solvespace/builds/202986194):
> #1446 passed
>Ran for 11 min 50 sec
>Total time 47 min 36 sec

Let's say 50min for one build. If credits=minutes per moth, that would be 200 builds per month and it may be OK for SolveSpace for now...

ppd, 2020-11-20:
OS | # Credits perstarted build minute
-- | --
Linux | 10
Experimental FreeBSD | 10
Windows | 20
MacOS | 50

ppd, 2020-11-20:
Well, it seems to be working as usual for the orga account at this time. I'm playing around with GH actions to have something prepared if the need arises. Fortunately, our CI workflow is rather straightforward and should translate easily; maybe except for the ARM64 builds, but I'll test that soon enough.

ruevs, 2020-11-20:
> macOS 31 min 36 sec
> Ubuntu 6 min 17 sec
> Windows 9 min 43 sec

So one build is about 1870 credits, or only 5 builds per month?!? :-(

ppd, 2020-11-20:
That seems to be the case.
For more, either pay or ask for free credits, possibly repeatedly.

The reasoning:

> However, in recent months we have encountered significant abuse of the intention of this offering (increased activity of cryptocurrency miners, TOR nodes operators etc.)

phkahler, 2020-11-20:
Let's yank flatbuffers, that increased build time quite a bit. I'm not convinced GH is a great solution either, but if we can't get enough free credits it will have to do.

Funny thing about CI. my computer sits idle most of the time. It would be fine to have it run CI but there's no way to have it test Windows, macOS, or ARM builds.

It seems OSS projects need a distributed multi-platform CI system that doesn't rely on 3rd party resources.

Symbian9, 2020-11-20:
Lets take a look on alternatives to ~~*Travis CI*~~:

- https://alternativeto.net/software/travis-ci/

What about *Buildkite CI*? It provide [Free Plan for FLOSS projects](https://buildkite.com/pricing) & has integration with GitHub, for example `rehex` project already uses it:

- https://github.com/solemnwarning/rehex
  - https://buildkite.com/solemnwarning/rehex/builds

vespakoen, 2020-11-20:
I also messed around with github actions for solvespace a while ago over here: https://github.com/vespakoen/solvespace/tree/gh-actions

I'd be happy to help with this, and also with universal macos builds.

ppd, 2020-11-21:
If there's the need to switch, a platform with a lot of resources, developers and popularity is the obvious choice if we want to keep it simple. That'd be GitHub Actions, especially considering we're already on GitHub.

I've ported our Travis workflow to GHA and will create a draft PR soon to invite further technical discussion. Whether the Travis situation will resolve itself or not, we'll still have another option then.

phkahler, 2020-11-21:
@ppd thanks for looking into it!  and @vespakoen help is much appreciated and I'd be really kind of excited to have a native ARM executable for macOS, it'd be great for the project.

ppd, 2020-11-23:
[image]

ruevs, 2020-11-23:
Github actions mentions 2000 free "build minutes" in my free github account. Still sounds better than the 10000 "credits" on Travis.

ppd, 2020-11-23:
I think that's 2000 free minutes for private repos and unlimited for public repos.

ruevs, 2020-11-24:
> [2,000 minutes/month<br/>Free for public repositories](https://github.com/pricing#feature-comparison)

I think you are right. It was not clear from the "Billing & lans" summary page.

ppd, 2020-11-24:
Time for a strategy decision. Which horse shall we bet on?

a) **Travis**. Actionable issue: Need to ask for OSS credits. How many? Is it a wise choice for the future aside from the cost/credits issue?
b) **GitHub Actions**. An implementation exists. See the remaining little issues in #824. Nothing actionable at the moment though.
c) **Something else**

As we're currently not doing any CI/CD, I'd prefer to decide on a strategy soon. @ruevs @phkahler @jwesthues @vespakoen @rpavlik, votes & opinions welcome.

Symbian9, 2020-11-24:
Agree on GitHub Actions (as for now), but for future need take a look on other options too (if something wild will happen with GitHub Actions, even GitHub iteself).

FTR, I recommend later to try created setup scripts for *Buildkite CI*, *GitLab CI* and other CI.

As for *GitLab CI*, it would be cool to create a mirror of SolveSpace repos on GitLab (it also could be reserve copy, less "hard-coded" on GitHub-only) & setup GitLab CI actions as it already did for GrafX2 & Inkscape:

- https://gitlab.com/GrafX2/grafX2/-/jobs
  - http://grafx2.gitlab.io/grafX2/
    - https://github.com/pulkomandy/grafx2
- https://gitlab.com/Inkscape/inkscape/-/jobs
  -  ~~http://inkscape.gitlab.io/inkscape/~~
    - ~~https://github.com/Inkscape/inkscape~~

ruevs, 2020-11-24:
My choice would be:
- Definitely add the GitHub Actions - they seem to be unlimited for free projects and should work.
- Remain on Travis and ask for credits - if it works - it works :-)

In this way (if Travis gave us credits) there would be two independent CI systems (as there used to be when Whitequark's AppVeyor was building Win32). In my opinion this is a very good thing. A few wasted CPU cycles - but they are "cheap".

With two CI systems running in parallel @ppd and @phkahler should decide which one will make the official releases.

No other CI systems and no GitLab in my opinion. I hate projects that are forked on all the OSS public servers and one never knows which one is the active/official one. And most of them "rot". If GitHub (Microsoft) were to go down, salvaging Solvespace from one of many private forks should be easy. The the git repository will not be lost. The big loss will be all the metadata - issues, pull requests, and the discussions attached to them.

ppd, 2020-11-24:
I think the scope of this issue is to get something stable running again quickly. Any mirroring and such goes beyond that (and I wouldn't be in favour of that anyway). And finally: this is not a CI playground project;  any option is a tool to help development of SolveSpace. Every service we use increases maintenance cost.

Using both is certainly an option. Redundancy has its advantages. Hmm.

jwesthues, 2020-11-24:
At @phkahler's ultimate discretion, but GitHub Actions seems like the obvious choice to me. Travis's new private equity owners probably see little business value in free software users, so I'd expect that to progressively tighten. No objection to duplicate CI, (though I'd try to avoid that myself), but no duplication of the repo, indeed confusing.

rpavlik, 2020-11-24:
GitHub actions is roughly based on Azure DevOps, which I've had great success with, it's much nicer than Travis and AppVeyor. Another vote for just switching to GitHub actions and not looking back. Any scripts that aren't environment specific should probably be separate from the CI config script for portability, in any case.  (For OpenXR stuff, I actually have the same docker images and nearly the same CI configs running internally on a private gitlab as well as publicly on azure devops - but I think SolveSpace is less demanding to build than an industry standards document and associated SDK.)

phkahler, 2020-11-24:
> At @phkahler's ultimate discretion, but GitHub Actions seems like the obvious choice to me. Travis's new private equity owners probably see little business value in free software users, so I'd [...]

And I will defer to @ppd on this stuff, so go with GH actions (seems to be a consensus anyway).

I'd also say try to keep it portable where it's not too much effort. GitHub is owned by MS and while I don't think they'll destroy it like [whomeveritwas] did to Source Forge, they are a BigCorp and have a history of acting like one.  To @ruevs point about losing metadata if we need to relocate the repo some day, that day will eventually come so yeah we should make solid commit messages that are as self contained as possible. Code comments are nice too if they give an explanation for why something is the way it is. There is also the developer_docs folder or whatever it's called within the repo which could use more content ( I keep meaning to pull some good descriptions from GH comments describing some internals ).

ppd, 2020-11-24:
I conclude that we agree on going the GitHub Actions route. Re the CI agnostic scripts: we mostly use the exact same scripts we used on Travis 1:1. So that's already good.

I'll polish the PR with the invaluable help of @vespakoen, and then we can land it soon.

jwesthues, 2020-11-24:
I think GitHub has an API for issues and such? If we were paranoid then we'd keep a continuously-updated private copy just in case, though we'll probably have sufficient notice to export if/when the time comes.

Symbian9, 2020-11-24:
> @ruevs: In this way (if Travis gave us credits) there would be two independent CI systems (as there used to be when Whitequark's AppVeyor was building Win32). In my opinion this is a very good thing. A few wasted CPU cycles - but they are "cheap".

Just my last drop on this thread, *LeoCAD*, for example, already do the same, where on each commit:

- Travis CI used for Linux/AppImage builds - https://github.com/leozide/leocad/blob/master/.travis.yml
- AppVeyor CI used for macOS & Windows builds - https://github.com/leozide/leocad/blob/master/appveyor.yml

Then all artifacts uploaded to GitHub as assets to `continuous` pre-release (such as `edge` in *SolveSpace*).

closed 2020-12-08 by vespakoen

Symbian9, 2020-12-08:
P.S.: ***"Travis CI is no longer providing CI minutes for open source projects"***

- https://news.ycombinator.com/item?id=25338983