solvespace/solvespace #808: Paste transformed / mirror (flip) is broken for text

calin, 2020-11-20:
### System information

SolveSpace version: 3.0~427a29ab

Operating system: Debian bullseye

### Expected behavior

Mirror text properly with _Paste transformed / mirror_.

### Actual behavior

As you can see in the attached screenshot, the triangles in the bottom quadrants (constructed from line segments) are mirrored properly, while the text object at the top is not, it's rather rotated in-plane and then translated. 

In addition, the scale is not preserved, the copy is smaller which would indicate more breakage.

[image]

ruevs, 2020-11-22:
SolveSpace does not allow the font object to be mirrored - no matter how one tries to constrain or contort it the text always remains "readable". It is intentional - whether it is a bug or a feature is a different question.

The size change on the other hand is interesting. If one constrains the size before the "Paste transformed..." then - of course - it retains it's size.

phkahler, 2020-11-23:
Valid issue.  The workaround may be to make a Rotate group around the vertical axis - 2 copies will be 180 degrees apart. That only works if it is the final step in a sketch though - it's not as flexible as paste-transformed.

I'm not sure why the text changes size, that's strange. @ruevs If I recall correctly a text entity has an associated normal and I want to say it shares the sketch normal. In order to copy it reflected we would need to make a copy of the normal and flip it. My mirror branch(es) is quite capable of mirroring text in both 2D and 3D but that creates a new group and needs to be completely reworked.

ruevs, 2020-11-23:
I've not gone into those parts of the code, sorry. My comment was based solely on using the current text entity implementation that does not allow it to be mirrored.

A rotate group does not mirror the text (or any objects).

calin, 2020-11-23:
IMO it is desirable to be able to easily create mirrored text, as an example for modeling casting molds or items with text that has to be looked at from the back.

One straightforward way to do this is to generate mirrored text right from the beginning, by implementing a "mirrored" text object property. One then can modify _Paste transformed / mirror_ to toggle the "mirrored" property in the copy when applied to text, thus improving workflow.

Of course the orthogonal way is to be able to apply arbitrary geometric transforms to arbitrary objects and groups, both destructively and non-destructively but that requires plenty of work.

phkahler, 2020-11-25:
[text_flip.zip](https://github.com/solvespace/solvespace/files/5593454/text_flip.zip)
[image]
Sketch the text
Select the origin and the little UP arrow at the origin, or whatever mirror axis you want.
Press SHIFT-R to create a rotate group around the Y axis.
Change it to 2 copies instead of the default 3. These will be coplanar and hence extrudable.
Use the menu to select "Sketch In Workplane" since it won't be in that mode.
Extrude.

This is a workaround to get mirroring. I still hope to have a look at the paste-transformed stuff to see if it can be done that way. Also still planning mirror groups for version 4.0 (some day).

@calin  You use SolveSpace for creating molds? Another thing I was messing with some time ago is adding draft angles to extrusions. I'm guessing you'd like that one quite a bit? If so and if you can build from source, you can check out my "draft" branch here:
https://github.com/phkahler/solvespace/tree/draft
It's a very old branch and has been over a year since I looked at it, but IIRC it adds a "draft angle" parameter in the text window for extrusions. It's probably also buggy and definitely not for production use. I haven't really got any feedback on it.

ruevs, 2020-11-25:
Hey that's great I did not think of rotating around the Y axis! :-)