solvespace/solvespace #809: BIM and scripting

JuanJ67m, 2020-11-21:
solvespace can be used for BIM?

exist a scripting option for Solvespace?

ruevs, 2020-11-21:
> solvespace can be used for BIM?

What is "BIM? Google offers many possibilities. [Building Information Modeling](https://knowledge.autodesk.com/support/revit-products/learn-explore/caas/video/youtube/lesson/143344-courseId-100332.html)? If yes, then no.

> exist a scripting option for Solvespace?

No. There is a [command line interface](https://github.com/solvespace/solvespace/issues/129#issuecomment-266179357) and a [Python wrapper of the solver](https://github.com/solvespace/solvespace/issues/292).

Symbian9, 2020-11-24:
> No. There is a command line interface

It's possible to provide some scripting by creating `*.sh`/`*.bat` script  (`grep` + maybe some other utilities would be required too) which would be wrapped around SolveSpace CLI aka `solvespace-cli`. I already though about creating such Shell-script for own needs, but any bundled scripting (maybe Lua-based?) would be really cool & is really needed feature for *SolveSpace* `3.0+`.

> > solvespace can be used for BIM?
>
> What is "BIM? Google offers many possibilities. Building Information Modeling? If yes, then no.

There is article on Wikipedia:

- https://en.wikipedia.org/wiki/Building_information_modeling

Also wide range of FLOSS apps (FreeCAD, Blender, ) already working on support *BIM Collaboration Format (BCF)*, see article on OSArch Wiki:

- https://wiki.osarch.org/index.php?title=BIM_Collaboration_Format_(BCF)
  - https://en.wikipedia.org/wiki/BIM_Collaboration_Format

As I said above, creating additional Shell-script wrapping for `solvespace-cli` + `grep` + `nano` (or any other CLI-based text editor + some other utilites) could be a workaround here; I already has an ideas on creating such and writing BOM info ([Bill of materials](https://en.wikipedia.org/wiki/Bill_of_materials)), EBOM ([Engineering bill of materials](https://en.wikipedia.org/wiki/Engineering_bill_of_materials)) & MBOM ([Manufacturing bill of materials](https://en.wikipedia.org/wiki/Manufacturing_bill_of_materials)) into `*.csv` by parsing groups names (which would include key words added by user via SolveSpace) from `*.slvs` file and some custom BOM template (something like `slvs-bom-catalog.csv`).

Suggest, creating some Shell-script for writing BIM into BFC format also could be solved by such workaround.

ruevs, 2020-11-26:
Since this is answered I'll close it. If at some point someone decides to work on these features they can continue the discussion here.

closed 2020-11-26 by ruevs