solvespace/solvespace #811: Units for Area inconsistent with chosen drawing units

Celloman15, 2020-11-21:
### System information

SolveSpace version: 3.0~bcb8cd2c

Operating system: <Win 10>

### Expected behavior

Select drawing units as mm. Analyse closed space area gives answer in mm2

### Actual behavior

Select drawing units as mm. Analyse closed space area gives answer in m2.  To get desired accuracy have then got to configure for many digits after decimal point which is not needed/desirable for the basic dimensions.

### Additional information

Example 2d example attached
[generic parametric guitar shape.zip](https://github.com/solvespace/solvespace/files/5577558/generic.parametric.guitar.shape.zip)

phkahler, 2020-11-21:
@Celloman15  To clarify you'd prefer something like 300 cm^2 instead of 0.03 m^2 correct?

I was looking into this and it seems the internal decision of printing km, m, mm etc...  is best for linear dimensions. The results are correct but the choice of when to switch units isn't great for area and is probably even worse for volume (0.001 cm3 to 999 cm3 are all in cubic cm).

Celloman15, 2020-11-22:
Hi phkahler,Yes, that's exactly it. The previous version would give areas in mm2 if dimensions were in mm which could lead to some pretty large numbers but at least you could see all the significant digits. Your suggestion to use cm2 is probably a better solution and definitely better than m2.ThanksDavid

phkahler, 2020-11-22:
> your suggestion to use cm2 is probably a better solution and definitely better than m2.

Well it already uses cm but the transitions from mm->cm->m are poorly chosen for area and volume. I've got a fix pending. Thanks for reporting the problem and the feedback!

phkahler, 2020-11-22:
@Celloman15  Since the fix is merged it should be available in the next "edge" build here:
https://github.com/solvespace/solvespace/releases

This fix hasn't made it out yet but should be there within an hour. I'll leave this issue open for a while longer in case you want to comment on the change either way - I do appreciate feedback on fixes.

closed 2020-11-24 by phkahler

Celloman15, 2020-11-26:
Downloaded the latest mp edge build today and re-tested on Win10.
On my example drawing (units set to mm with "0 digits after decimal places on distances")  the area shows up in m2 but with 6 digits after decimal point so this is ok.  If a change to "1 digit after decimal places on distances" the area shows up with only 2 digits after decimal point which is not helpful.

ruevs, 2020-11-26:
The edge build does not contain the change you need. The continuous integration is temporarily down.
Try this one if you want https://github.com/solvespace/solvespace/issues/817#issuecomment-733970990

For me. If "digits after decimal point to show distances"= 0 the result is:

> The area of the region sketched in this group is:
> 292 cm²

If 3 digits then

> The area of the region sketched in this group is:
> 292.071 cm²

and so on.

Celloman15, 2020-11-27:
@ruevs That version works perfectly thanks.