solvespace/solvespace #813: Suggestion: Use feet for imperial units with large values

phkahler, 2020-11-22:
The largest imperial unit of measure in SolveSpace is currently the Inch. For long lengths it would be good to switch the displayed units to feet.  For area square feet, and volume cubic feet.. These would only be for display, all inputs would still be in inches.

At what length, area, and volume should the displayed values change to feet?

Indoor dimensions are often in inches (I don't want to do mixed feet/in but anything is possible) so the change should be somewhere between 10 and 20 feet?  For area I'd guess more than a few hundred sq inches should be in sq feet? For volume perhaps change to cubic feet after 1000 in^3 which is about 0.58 cubic feet.

ruevs, 2020-11-22:
Mmm let's not forget [cubic shaftments](https://en.wikipedia.org/wiki/English_units#Length) ;-)

phkahler, 2020-11-22:
@ruevs  You may laugh, but in the US most home construction is done in feet and inches (and fractions of an inch). Putting windows in your house?  Width and height in inches. New flooring? square feet.  Sidewalk work? Length in feet.  A refrigerator may well indicate volume in cubic feet. Getting a load of dirt, gravel, or crushed stone?  Order that in cubic yards.

ruevs, 2020-11-23:
I know, but never got used to it. Sorry for the inappropriate joke.

phkahler, 2020-11-23:
> I know, but never got used to it. Sorry for the inappropriate joke.

Oh no. Joke was fine! I think our situation with units is kind of absurd ;-) I need to use more emoji.  It's really absurd since there was a major push to switch to SI as long ago as 1980 and we are still not there ;-(  it was really sad when NASA crashed a probe into Mars due to failure to convert units in the 90s. But I digress...  :-)

Symbian9, 2021-08-07:
As SolveSpace has *"Dimensions in Meters"* and  *"Dimensions in Meters"* for `SI` system of measurements, it would be good if there would ***"Dimensions in Feets"*** additionally to *"Dimensions in Inches"* for `Imperial` system of measurement too.

Also it would be cool if there would be a way to show Imperial dimensions with `ft`/`in` and alternatively with `'`/`"`:

- in Feet:
  - 2.67ft
  - 2' 8"
- in Inches:
  - 32in
  - 32"

phkahler, 2023-08-10:
Not sure why this wasn't closed when #1088 was merged.

closed 2023-08-10 by phkahler