solvespace/solvespace #816: Remove CLA signing requirement for contributing

Symbian9, 2020-11-22:
### System information

SolveSpace version: `3.x`, `master`

Operating system: all supported

### Expected behavior

SolveSpace `3.x` should be fully free of CLA as SolveSpace `2.0` was released clearly under terms of `GNU GPLv3`

- http://libregraphicsworld.org/blog/entry/solvespace-released-under-gpl

Removing CLA signing (rolling back to only `GNU GPLv3`) is NOT breakage change because it is allowed directly by CLA signers who contributed earlier.

### Actual behavior

Signing CLA requirement was added by ex-maintainer few years ago (due to double-licensing for ex-maintainer needs) 

- https://github.com/solvespace/solvespace/commit/b2418a0324fa4e4ed498a46aa0946b67a267ebf0#diff-eca12c0a30e25b4b46522ebf89465a03ba72a03f540796c979137931d8f92055

Actually there are NO needs in CLA signing anymore for the future development of *SolveSpace* `3.x` under `GNU GPLv3`.

### Additional information

ruevs, 2020-11-22:
@Symbian9 this was already discussed see this https://github.com/solvespace/solvespace/issues/739#issuecomment-711052179 and the following three comments. The CLA signing does not appear for new contributors.

phkahler, 2020-11-22:
Feel free to make pull requests, the CI system is not checking the CLA and if does complain we'll merge good contributions anyway - this has already happened so the code does not have full CLA coverage to be distributed as anything but GPLv3 by anyone at this time.

@ruevs I think the notice may still appear, but the CI tests are not checking for it any more ;-)

kbrock, 2020-11-28:
Is it possible to just say that all future contributions are under GPL3 and MIT/BSD/Apache/LGPL (pick your poison)?

So future contributors understand the intent?

It is a lot of work to track down contributors to add a license retroactively, so best to not dig yourself deeper.

jwesthues, 2020-11-28:
@kbrock I believe such an approach would be legally sound; but in practice, I think the effort to disentangle the existing codebase would be big enough that it's very unlikely to ever happen. So I'm inclined to avoid that complexity and extra barrier to contributing, and just let SolveSpace be GPLv3-only.

Symbian9, 2020-11-28:
> let SolveSpace be GPLv3-only.

Fully agree! So may revert https://github.com/solvespace/solvespace/commit/b2418a0324fa4e4ed498a46aa0946b67a267ebf0#diff-eca12c0a30e25b4b46522ebf89465a03ba72a03f540796c979137931d8f92055 now?

Mention of CLA should be removed from `CONTRIBUTING.md`.

closed 2020-12-13 by phkahler

Symbian9, 2020-12-19:
@phkahler, CLA Assistant still linked to SolveSpace repo!

- https://cla-assistant.io/solvespace/solvespace

Need fully remove SolveSpace from CLA Assistant website, becase CLA Agreement text actually is fully violate to SolveSpace GPLv3 License!

>>> 

```
This document contains both the Individual Contributor License Agreement and Entity Contributor License Agreement. Only one of these agreements applies; please sign the Invididual CLA on behalf of yourself, or Entity CLA on behalf of your employer.
SolveSpace Individual Contributor License Agreement

Thank you for your interest in contributing to SolveSpace ("We" or "Us"). SolveSpace is developed by two legal entities, M-Labs Ltd, a Hong Kong limited liability company, and Useful Subset, LLC, a Washington limited liability company. All licenses granted below are granted to each entity, and each entity may use the licenses at its individual discretion.

This contributor agreement ("Agreement") documents the rights granted by contributors to Us. To make this document effective, please sign it and send it to Us by electronic submission, following the instructions at https://cla-assistant.io/solvespace/solvespace. This is a legally binding document, so please read it carefully before agreeing to it. The Agreement may cover more than one software project managed by Us.
1. Definitions

"You" means the individual who Submits a Contribution to Us.

"Contribution" means any work of authorship that is Submitted by You to Us in which You own or assert ownership of the Copyright. If You do not own the Copyright in the entire work of authorship, please contact support@solvespace.com by email.

"Copyright" means all rights protecting works of authorship owned or controlled by You, including copyright, moral and neighboring rights, as appropriate, for the full term of their existence including any extensions by You.

"Material" means the work of authorship which is made available by Us to third parties. When this Agreement covers more than one software project, the Material means the work of authorship to which the Contribution was Submitted. After You Submit the Contribution, it may be included in the Material.

"Submit" means any form of electronic, verbal, or written communication sent to Us or our representatives, including but not limited to electronic mailing lists, source code control systems, and issue tracking systems that are managed by, or on behalf of, Us for the purpose of discussing and improving the Material, but excluding communication that is conspicuously marked or otherwise designated in writing by You as "Not a Contribution."

"Submission Date" means the date on which You Submit a Contribution to Us.

"Effective Date" means the date You execute this Agreement or the date You first Submit a Contribution to Us, whichever is earlier.

"Media" means any portion of a Contribution which is not software.
2. Grant of Rights

2.1 Copyright License

(a) You retain ownership of the Copyright in Your Contribution and have the same rights to use or license the Contribution which You would have had without entering into the Agreement.

(b) To the maximum extent permitted by the relevant law, You grant to Us a perpetual, worldwide, non-exclusive, transferable, royalty-free, irrevocable license under the Copyright covering the Contribution, with the right to sublicense such rights through multiple tiers of sublicensees, to reproduce, modify, display, perform and distribute the Contribution as part of the Material; provided that this license is conditioned upon compliance with Section 2.3.

2.2 Patent License

For patent claims including, without limitation, method, process, and apparatus claims which You own, control or have the right to grant, now or in the future, You grant to Us a perpetual, worldwide, non-exclusive, transferable, royalty-free, irrevocable patent license, with the right to sublicense these rights to multiple tiers of sublicensees, to make, have made, use, sell, offer for sale, import and otherwise transfer the Contribution and the Contribution in combination with the Material (and portions of such combination). This license is granted only to the extent that the exercise of the licensed rights infringes such patent claims; and provided that this license is conditioned upon compliance with Section 2.3.

2.3 Outbound License

Based on the grant of rights in Sections 2.1 and 2.2, if We include Your Contribution in a Material, We may license the Contribution under any license, including copyleft, permissive, commercial, or proprietary licenses. As a condition on the exercise of this right, We agree to also license the Contribution under the terms of the license or licenses which We are using for the Material on the Submission Date.

2.4 Moral Rights. If moral rights apply to the Contribution, to the maximum extent permitted by law, You waive and agree not to assert such moral rights against Us or our successors in interest, or any of our licensees, either direct or indirect.

2.5 Our Rights. You acknowledge that We are not obligated to use Your Contribution as part of the Material and may decide to include any Contribution We consider appropriate.

2.6 Reservation of Rights. Any rights not expressly licensed under this section are expressly reserved by You.
3. Agreement

You confirm that:

(a) You have the legal authority to enter into this Agreement.

[... cut ...]
6.5 If any provision of this Agreement is found void and unenforceable, such provision will be replaced to the extent possible with a provision that comes closest to the meaning of the original provision and which is enforceable. The terms and conditions set forth in this Agreement shall apply notwithstanding any failure of essential purpose of this Agreement or any limited remedy to the maximum extent possible under law.

Us:

    Useful Subset LLC, located at 818 SW 3rd Ave #221-5721, Portland OR 97204, United States
    M-Labs Ltd, located at Workshop 15B, 6/F, Block B1, 17 Ko Fai Rd, Yau Tong, Hong Kong

Harmony (HA-CLA-E-ANY) Version 1.0
```

<<<