solvespace/solvespace #817: Crashing when calling extrude on two circles

NothanUmber, 2020-11-23:
### System information

SolveSpace version: 3.0~bcb8cd2c

Operating system: Windows 10 1909

### Expected behavior

a face is drawn between two circles when using extrude

### Actual behavior

the application crashes with error message

---------------------------
Fatal error â€” SolveSpace
---------------------------
File C:\Users\travis\build\solvespace\solvespace\src\platform\guiwin.cpp, line 1338, function SolveSpace::Platform::WindowImplWin32::ConfigureScrollbar:
Win32 API call failed: SetScrollInfo(hWindow, SB_VERT, &si, TRUE).
Error: Das System kann die angegebene Datei nicht finden. (Translation: The system cannot find the indicated file)

### Additional information

steps to reproduce:
* draw two circles around the same center point with slightly different radius (e.g. 2.0 and 2.2 cm)
* select both circles
* use the "extrude with new active sketch" tool

I am completely new to this software and tried to create a surface between two circles that I then wanted to further extrude to get a tube with a certain thickness. Not really sure whether the extrude option I chose is the right thing to create the face between the two circles. But it probably shouldn't crash.
Was able to reproduce that with both the OpenMP and the normal Windown build.

ruevs, 2020-11-23:
So you are trying to create this model?
[pipe_ID20_OD22.zip](https://github.com/solvespace/solvespace/files/5580742/pipe_ID20_OD22.zip)

- What is visible in the "Property browser" when this happens?
- Does it crash if you just open the attached model?
- Can you reproduce it consistently?
- Can you reproduce it in a different way - e.g. a "Lathe" (`Shift-L`)  instead of extrude?

NothanUmber, 2020-11-23:
Yes, that was what I tried to create.
1) see screenshot
[image]

2) no, the attached model works fine
3) yes
4) when pressing Shift-L with both circles selected I get the following message - no crash:
[image]

NothanUmber, 2020-11-23:
P.S.: Further experiment: It already crashes with the same error message when just clicking on one of the two circles with the left mouse. (Previously I selected the circles by drawing a box around them). And when I try to open e.g. the Edit-Configuration dialog. Always with the same error message shown above.
So there might be a special incompatibility with my system? Tried to start the application with both User and Admin rights, didn't make a difference. I use a 4k monitor with 125% text scaling, 32 GB RAM, Nvidia 1080 TI GPU, Ryzen 1700 CPU.

NothanUmber, 2020-11-23:
I found out what the problem is: Solvespace and Display Fusion (https://www.displayfusion.com/) seem to be incompatible. (DisplayFusion is a solution that can be used to e.g. manage several windows on a bigger monitor, multi monitor setups etc.)
When stopping DisplayFusion the problem disappears (and e.g. Extrusion, object selection and opening the Configuration dialog all work). When starting DisplayFusion again, then Solvespace crashes again. Using DisplayFusion Pro 9.7.

phkahler, 2020-11-24:
@NothanUmber  Some new user stuff:  you don't need to select anything to do an extrude, just click the icon or press shift-X or use the menu.  Lathe, Revolve, and Helix all require that you select an axis to rotate around - this should be a point and normal vector, or a point and line segment (use a construction line to make a donut). But you don't need to select the items to rotate - it will apply to the whole sketch.  It will also tell you via popup what selection is needed once you get past the crash on those operations if you didn't select the right things.

ruevs, 2020-11-24:
This is a well hidden bug in SolveSpace somehow exposed by Display Fusion.
It crashes here:
https://github.com/solvespace/solvespace/blob/f2850246fa4f56c9ef080d620d9b7266764dd273/src/platform/guiwin.cpp#L1345
Which is in fact an assert check that should not be done since [SetScrollInfo returns the scrollbar position](https://docs.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-setscrollinfo?f1url=%3FappId%3DDev16IDEF1%26l%3DEN-US%26k%3Dk(WINUSER%252FSetScrollInfo);k(SetScrollInfo);k(DevLang-C%252B%252B);k(TargetOS-Windows)%26rd%3Dtrue#return-value) and not a status flag.

However `sscheck` calls [`CheckLastError`](https://github.com/solvespace/solvespace/blob/f2850246fa4f56c9ef080d620d9b7266764dd273/src/platform/guiwin.cpp#L73), which in turns calls [GetLastError](https://docs.microsoft.com/en-us/windows/win32/api/errhandlingapi/nf-errhandlingapi-getlasterror) and since normally there is no error nothing bad happens. But apparently when running with Display Fusion there is a "last error" for some reason and it crashes.

ruevs, 2020-11-24:
@NothanUmber please try if the attached version works with Display Fusion for you.
[solvespace_PR827_Win32_x86_GL1.zip](https://github.com/solvespace/solvespace/files/5591733/solvespace_PR827_Win32_x86_GL1.zip)
(the continuous integration is currently not running )

NothanUmber, 2020-11-24:
@ruevs Had a look, works now with DisplayFusion enabled. Thanks a lot!

@phkahler Ah, ok, thanks! Am still only scratching on the surface and am currently going through some Youtube tutorials. I really like the minimal UI philosphy and focus of SolveSpace so far!

NothanUmber, 2020-11-24:
Nooo, another crash (with DisplayFusion enabled), with a new error message this time:

---------------------------
Fatal error â€” SolveSpace
---------------------------
File C:\Users\ruevsk01\Desktop\solvespace\git\src\platform\guiwin.cpp, line 1319, function SolveSpace::Platform::WindowImplWin32::ShowEditor:
Win32 API call failed: ShowWindow(hEditor, SW_SHOW).
Error: Das System kann die angegebene Datei nicht finden.

NothanUmber, 2020-11-24:
Yes, think I got it reproducible now: Double click on the right side distance constraint and try to change it from 11.something to 2. Crashes reproducibly for me when DisplayFusion is enabled and works without problems with DisplayFusion disabled.

[flöt2.zip](https://github.com/solvespace/solvespace/files/5592586/flot2.zip)

NothanUmber, 2020-11-24:
Seems to be this line: sscheck(ShowWindow(hEditor, SW_SHOW));
Looks right on first sight? According to the docu ShowWindow returns a BOOL, as expected from sscheck? https://docs.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-showwindow

ruevs, 2020-11-24:
Yup, even though I reviewed all `sscheck` calls, this is one that I missed - because it returns BOOL, but it can be false for perfectly valid reasons.
https://docs.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-showwindow#return-value

ruevs, 2020-11-24:
> Seems to be this line: sscheck(ShowWindow(hEditor, SW_SHOW));
> Looks right on first sight? According to the docu ShowWindow returns a BOOL, as expected from sscheck? https://docs.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-showwindow

Our posts crossed :-)
Yes it returns BOOL, but not as an error flag:
> If the window was previously hidden, the return value is zero

But this is exactly how I missed it - since there were many calls to check after some point I got lazy and started treating functions returning BOOL ad fine without reading the documentation :-)

ruevs, 2020-11-25:
@NothanUmber try the attached version.

However after looking much more carefully at he code I can fairly confidently say that DisplayFusion does something bad/buggy. Probably after injecting itself into each process - to achieve it's... windows management? functions - it makes bad Win32 calls that cause GetLastError to be non zero.

If I were you I would file a bug report with DisplayFusion and point them to this issue and the linked pull requests. It would be interesting how they would react.

[solvespace_PR817_Win32_x86_GL1.zip](https://github.com/solvespace/solvespace/files/5598597/solvespace_PR817_Win32_x86_GL1.zip)

NothanUmber, 2020-11-25:
@ruevs Thank you for looking into it again! Good idea, will open a DisplayFusion ticket, might be worth looking at from their side.
SolveSpace is still crashing for me with the same error message (and the same line number) when using the new build.

NothanUmber, 2020-11-25:
Posted in their forums, they are usually quite responsive. Let's hope they find something!
https://www.displayfusion.com/Discussions/View/displayfusion-can-crash-solvespace/?ID=0b3c924c-7594-4157-bd4b-17af480401f6#2

closed 2020-11-25 by phkahler

ruevs, 2020-11-25:
@NothanUmber Hmm I zipped up the wrong exe - was juggling too many branches...

Try one more time with this one please.

[solvespace_PR817_OneMoreTry_Win32_x86_GL1.zip](https://github.com/solvespace/solvespace/files/5599483/solvespace_PR817_OneMoreTry_Win32_x86_GL1.zip)

Help | About should say:
> This is SolveSpace version 3.0~ a2b5d0d4.

NothanUmber, 2020-11-25:
@ruevs  Yay, works, haven't found any more problems in a short test, great! Thanks a lot again!

NothanUmber, 2020-12-15:
The problem seems to be fixed now in DisplayFusion 9.7.1 Beta 9. Great!
Think we can close this ticket.

ruevs, 2020-12-16:
Your post in the DisplayFusion forum was very clear and I am glad they reacted and fixed it. It speaks well of them and confirms my theory. Thus there are no open questions hanging.

Thanks for following up on this. I had left the forum thread open in a tab, but had not looked at it in since your "Disabling application hooks seems to have helped also." reply :-)