solvespace/solvespace #819: [3.0rc1] adding a line segment results in status "err" for "degree of freedom"

fe60, 2020-11-23:
### System information
SolveSpace version: 3.0.rc1
Operating system: Windows 10 (2004)

### Bug
After adding a line segment, the status shown for **"degree of freedom" switches from "ok" to "err"**.
Remove the line segment and the status switches back to "ok".
The problem does not occur with SolveSpace 2.3.
[image]

### Additional information
slvs-file shown in the screenshot
[dof_err.zip](https://github.com/solvespace/solvespace/files/5581666/dof_err.zip)

ruevs, 2020-11-23:
The "err" indicator is a new feature. It is intended to indicate that the sketch would not result in a valid solid if extruded for example. The feature was implemented here c2c26e95adb and last changed here df6777aaf39.

There is a small problem with it, that at the end of adding a closed polygon the status remains on "err" until `Esc` is pressed or a left click anywhere.

ruevs, 2020-11-23:
I added a description in the [changelog](https://github.com/solvespace/solvespace/commit/8dcb2db6e6b9539e82b7ae8b0fe967914b5f90fc).

fe60, 2020-11-23:
@ruevs  I use SolveSpace for 2d modeling only. Hence I have deactivated the option "check sketch for closed contour".
From my point of view this should deactivate the newly introduced warnings and errors in the list of groups too.

ruevs, 2020-11-23:
@fe60 this makes sense to me and is easy to implement. I created a pull request. Lets see what the others and @wpwrak (who implemented the feature) think.

phkahler, 2020-11-23:
@ruevs  Is the check enabled by default on a fresh install?  I think these new options should probably be enabled by default - things like automatic constraints and "edit newly added dimensions". If not enabled, new users and even people upgrading may never notice them or realize what the option does. Existing users who don't want the new features will get them and perhaps complain about new behaviour at which point we can say - turn it off here. Does that make sense?

ruevs, 2020-11-24:
It makes perfect sense.
It is turned on by default:
https://github.com/solvespace/solvespace/blob/f2850246fa4f56c9ef080d620d9b7266764dd273/src/solvespace.cpp#L82

closed 2020-11-24 by phkahler