solvespace/solvespace #826: [3.0.rc1] Enable "edit newly added dimensions" by default

fe60, 2020-11-24:
SolveSpace version: 3.0.rc1
Operating system: Windows 10 (2004)

I'm new to Solvespace, but one thing I wanted to get rid of pretty soon, was the necessity of double clicking a newly added dimension. Luckily I started with the latest release candidate and found the "edit newly added dimensions" checkbox.

I strongly recommend to enable this option by default to make people aware of it. I guess only a minority will be bothered and switch it off.
https://github.com/solvespace/solvespace/blob/f2850246fa4f56c9ef080d620d9b7266764dd273/src/solvespace.cpp#L80

ruevs, 2020-11-24:
This was discussed when the feature was initially implemented.
https://github.com/solvespace/solvespace/pull/514#issuecomment-558753794
and the decision was to not change the existing behaviour  by default.

But the feature is indeed very useful. Perhaps we can add a more prominent mention of it - for example in the ["Quick start" section of the download page](https://solvespace.com/download.pl)?

phkahler, 2020-11-24:
I am very much in favor of this change and specifically mentioned it in a comment yesterday (which I can't find). The reasoning is simple, new users or even old users will not have to discover new features if we turn them on by default. Current users who don't like a new behavior may complain, and we can tell them where to turn it off. They are also more likely to be familiar with the config screen and just might find it themselves.  This recommendation is only for options like this that IMHO make using SolveSpace easier. 

As a counter example the original implementation of automatic constraints was a very mixed bag for me due to the automatic over-constraining of sketches that happened too often. I wouldn't have recommended that be the default but since it has improved so much - redundant constraints are often not added - I am in favor of having it by default.

ruevs, 2020-11-24:
Fine with me. I'm ambivalent on the matter. Just linked to the existing discussion.

closed 2020-11-25 by ruevs