solvespace/solvespace #831: Review terms in translation source

Symbian9, 2020-11-30:
### System information

SolveSpace version: [`3.0~94b26dd`](https://github.com/solvespace/solvespace/commit/9390ab02d5e1d921cfaa4e4948082a818d0e8246)

Operating system: all supported

### Expected behavior

In *SolveSpace* `3.0` terms should be finally corrected to follow common CAD Terminology (such as used in FreeCAD, QCAD/LibreCAD).

- Replace `2d`/`3d` terms with `2D`/`3D`.
- Set first letter capitalized in all Terms/Words where it would improve readability, especially on "Property Browser", where mostly all strings are lowercase-only — is really hard to read.
- Remove double meaning of "`... could not be...`" and "`... must not be...`" — use only "`... could not be...`" instead.
- ***TBA*** more soon (I will add more examples soon, as I'm working on translation & may discover more issues)

> **NOTE:** After UI translation source (locale template) would be fixed, then docs should be corrected too

### Actual behavior

There are a lot of terms & definitions in *SolveSpace* `master` which are not correspond to the accepted terminology in CAD software.

### Additional information

### REFERENCES

- http://people.virginia.edu/~rlk3p/classes/dram718/Reference/Glossary/Terms.html

phkahler, 2020-12-02:
I don't want to update the translation source right now. All of the translations are mostly up to date and we may not have people to update them again prior to 3.0.

As for the text window... it has a consistent style being all lower case and changing it will require code changes all over the place so I don't want to do that now either. It's not clear that introducing mixed case there will help readability either. The only thing I'd really be willing to change right now is 3d->3D and 2d->2D in the text window where it doesn't require translation updates and is very minimal.

The text window could benefit from a lot of rework way beyond string changes, but that's got to wait until a more comprehensive plan and people show up to implement it.

Symbian9, 2020-12-02:
> The only thing I'd really be willing to change right now is 3d->3D and 2d->2D in the text window where it doesn't require translation updates and is very minimal.

Agreed.

> we may not have people to update them again prior to `3.0`.

Okay, lets leave this task for next `3.x` releases.