solvespace/solvespace #832: Node Editing Mode

giannissc, 2020-11-30:
Would a node editing mode ala Blender or [Antimony](http://www.mattkeeter.com/projects/antimony/3/) fit into the roadmap for solvespace?

phkahler, 2020-11-30:
@giannissc what is the definition of "node" in this request? Is it a form of direct manipulation of the NURBS surfaces? Or more importantly what would it be used for?

giannissc, 2020-11-30:
Node editing is a style of visual programming where a node can be anything (2D object, 3D object or an operation on those objects such as extrusion). I think it is a very intuitive way of working especially for beginners. If you have a look at the link i posted earlier it shows an node ediiting workflow in practice.

Link: https://vimeo.com/125111378

giannissc, 2020-11-30:
I was look for simple CAD programs for linux today and I stumbled upon this project as well as Antimony. I just think a combination of both programs would be a really powerful one! Just didn't know if you were familiar with Antimony and thought it might be something that you like as well.

phkahler, 2020-11-30:
> Would a node editing mode ala Blender or [Antimony](http://www.mattkeeter.com/projects/antimony/3/) fit into the roadmap for solvespace?

No, I don't think so.  Antimony and OpenSCAD are two different approaches to designing 3D objects. The philosophy behind SolveSpace (if there is one) seems to be "Draw it right on the sketch" along with "Use the constraint solver".  There may be some similarity between our series of groups and such a graph, but ours is very linear and we don't currently have the internal flexibility to support an arbitrary graph all at once - though you can emulate it by linking sketches.

Most of the non-commercial "CAD" programs lack basic functionality. Take OpenSCAD and probably Antimony and try to construct a triangle with side-lengths of 7, 8, and 9.  I don't think it will be easy and neither a directed graph nor a built in scripting language is going to change that.  I'm not even sure if SketchUp can do it.

giannissc, 2020-12-01:
Well I am a new user to CAD and SolveSpace, so I am misunderstanding how everything it supposed to work. Thanks for the reply :)

closed 2020-12-01 by giannissc

phkahler, 2020-12-01:
@giannissc it was a good question. Made me think about what it would take to even enable us to use graphs like that, and there is some overlap with other issues we have.

giannissc, 2020-12-03:
Well in that case happy to help :)
I have been playing around with solvespace these past couple of days and I got to admit it is really growing on me. It is a great program!
I was wondering though what is the relationship with NoteCAD. Does it use solvespace geometry kernel but a different interface @phkahler?

phkahler, 2020-12-04:
@giannissc   I'm not sure about NoteCAD.  My understanding is that it was a port/rewrite or simply inspired by some portions of SolveSpace written in C#.  Why that was done I never understood, but it's creator has been a long time SolveSpace user and contributor.  NoteCAD can not export STEP files I think because it doesn't use NURBS representations of surfaces. Maybe @Evil-Spirit will pop in and give us a little history on how it came to be?

As far as I can tell every open source program besides SolveSpace that handles STEP files or trimmed NURBS surfaces uses the OpenCascade (OCCT) library which is a complete (and huge) geometric kernel with many capabilities. The web site says a full installation with all the documentation takes 1.8GB.  That's over 100x bigger than SolveSpace. Here the geometry code is only about 10 files averaging 1000 lines each. The constraint solver is another few files, graphics is a few more and so on.  SolveSpace is minimal but quite capable. Have fun!

ruevs, 2020-12-04:
Ideed the "use OpenCSCADE" idea is as old as SolveSpace being open source.
https://solvespace.com/forum.pl?action=viewthread&parent=878&tt=1439203135

It usually dies very quickly, and for that I am very thankful. That codebase is an awful mess - even from my superficial browsing. Not to mention the size...

https://github.com/solvespace/solvespace/issues?q=opencascade

https://www.google.com/search?q=site%3Asolvespace.com%20opencascade