solvespace/solvespace #834: Add option to "Do not shadow inactive group"

Symbian9, 2020-12-01:
### System information

- SolveSpace version: `3.0~` https://github.com/solvespace/solvespace/commit/9390ab02d5e1d921cfaa4e4948082a818d0e8246
- Operating system: all supported

### Expected behavior

Add option to set *"Do not shadow inactive groups"* which would prevent "shadowing" of all solids in model/assembly.

### Actual behavior

Actually adding "Sketch in 3D" or "Sketch in Workplane" new group just after `lathe`/`extrude`/`translate`/`rotate` groups, applies `s003-#def-inactive-grp` color style to all previous solids/entities (e.g. all solids are "shadowed").

### Additional information

ruevs, 2020-12-01:
All 3D groups are "shadowed" when a 2D ("sketch-in-plane") group is active. When a 3D group is active the inactive 3D groups are drawn "normally". In my opinion this makes perfect sense. What is the use case for such an option?

Symbian9, 2020-12-01:
> What is the use case for such an option?

I use adding *"New Group > Sketch in 3D"* + enabling *"allow redundant constraints"* to draw measurements in 3D just next after solid group.

Video (showcase) — https://www.youtube.com/watch?v=1uZgUpjTg18

Screenshot:

[image]

Just for now (as a workaround; "wrong way") I should tune each time `s003-#def-inactive-grp` colors style (colors, lines, stipples, etc.) to be the same as last solid group colors; BUT it still not work for multi-color parts as "shadowing" fully "broke" multi-color parts for this usage (as it just set all colors to grey with default `s003-#def-inactive-grp`) 

> In my opinion this makes perfect sense.

I request just add *"Do not shadow inactive groups"* as *togglable option* (not replace existing behavior, but make it customizable instead) because it would be really useful for producing such types of drawings (exporting as 2D View) —  as for me this is fully ***must have*** feature for *SolveSpace* `3.0`.

phkahler, 2020-12-02:
@Symbian9  I was wondering how you made those drawings, they look very good. To clarify, after creating the entire object you add a new "sketch in 3d" group that only contains the dimensions, and you allow redundant constraints so they don't appear as reference. Having the constraints in their own group allows you to turn off all entities in previous groups which results in consistent and clean lines, while showing only dimensions from the last group. Correct? I realize you already said this using different (and fewer) words.

So the problem is that last group is treated as a sketch and shadows everything, but you'd like an option to preserve the colors instead of shadow them.

I'll think about that. It may be exactly what you say (an option that disables a behaviour), or it may be something a little different internally with the same visual appearance. I'm not sure we have a choice, but would this be best as a per-group option or a global configuration setting?

BTW another reason for creating a sketch in 3D with no new solid parts is to trace existing entites (points and maybe lines) for a part that will be used in an assembly. That last group can contain only the entities you want visible in the assembly with all others turned off to eliminate clutter.

Symbian9, 2020-12-02:
Here is `slvs` of this 3D piece (bellow added screenshots created with default SolveSpace colors styles):

- https://github.com/Symbian9/SolveSpace-Daily-Engineering/blob/master/cadtober/cadtober2020-day2.slvs

NOTE: Changing constraints dimensions in previous groups immediately make related redundant constrain in `g00e-sketch-in-3d` groups unsolvable — but it is very easy to update & solve it: just go to each unsatisfied constraint in `g00e-sketch-in-3d` via "Property Browser" and toggle `reference` property OFF (also need update canvas view using <kbd>Space</kbd> button) and then toggle it ON again. That's it!

1. [image]
2. [image]
3. [image]
4. [image]
5. [image]

>  Having the constraints in their own group allows you to turn off all entities in previous groups which results in consistent and clean lines, while showing only dimensions from the last group. Correct?

Yes and I use this trick to draw (if I would try to draw measurements just in the last solid group — there would be a lot of unneeded constraints & entities shown, which I can't disable to show because it will also hide all dimension measurements which I want keep to draw)

- [image]
- [image]
- [image]

> but would this be best as a per-group option or a global configuration setting?

As for me, at least global configuration setting should be added to toggle ON/OF "shadowing" of previous groups. (in *"Property Browser > line styles"* OR in *"Property Browser > configuration"*)

I'm not sure what you mean under a *per-group option — may you add your POV on this feature request?

Symbian9, 2020-12-02:
>  I was wondering how you made those drawings, they look very good

JFTR, Here is my new SolveSpace 3.0 showcase:

- https://www.youtube.com/watch?v=KY95-xrbSNo&list=PL53gPqUkbj3JndwS2T410KTbBymaimbiD
- SLVS: `30daycadchallenge-intro.slvs` - https://github.com/Symbian9/SolveSpace-Daily-Engineering/tree/master/30DayCADChallenge

phkahler, 2020-12-06:
@Symbian9  you make this look like a professional CAD tool.  You're choice of shading is really nice, did you notice we implemented the flat-shading you asked for? Just turn the lights down to 0 brightness and bring ambient up to 1.0 should do it, but I think this shading you're using is better than that.

Symbian9, 2020-12-06:
> did you notice we implemented the flat-shading you asked for? Just turn the lights down to 0 brightness and bring ambient up to 1.0 should do it

@phkahler Yeah, I'm aware of this trick & already playing with it a lot — I would record next showcase videos especially to describe all my "shading" tips'n'tricks techniques ;)

BTW, This issue/feature request has a little different behavior, and to make *SolveSpace* `3.x` looks more professional *"Do not shadow inactive groups"* is really 'must have' feature.

FTR, Additionally I has few more *ideas/feature requests* related to improve dimensioning/measuring solids in 3D. (would create separate issue per each feature request) 

P.S. Also, I already *found few more issues/bugs* (especially in export to SVG & DXF, which mostly related to dimension text labels and line stiles) — would report this week in separate issue threads.

phkahler, 2020-12-06:
> BTW, This issue/feature request has a little different behavior, and to make _SolveSpace_ `3.x` looks more professional _"Do not shadow inactive groups"_ is really 'must have' feature.
> 
I know, I heard that the first time you said it ;-)  I'm still hoping to have a look.

> FTR, Additionally I has few more _ideas/feature requests_ related to improve dimensioning/measuring solids in 3D. (would create separate issue per each feature request)
> 
An issue per feature is a good, then each can be discussed separately.

There are still several open related to line styles which I moved to the 4.0 milestone. I looked at several that appear to be Z-buffer related and had to stop. I'm good with geometry and many aspects of graphics, but I using openGL to draw lines along with 3D surfaces is apparently a very tricky thing to do well.

phkahler, 2020-12-12:
Should this be in the configuration screen, or the view menu? I'm leaning toward the view menu right under Show Snap Grid and above Use Perspective Projection. Maybe call it "Show solid model while sketching" or similar?

Symbian9, 2020-12-13:
> Maybe call it "Show solid model while sketching" or similar?

I propose use *"Fade previous solid groups"* (which by default should be enabled, as actual *SolveSpace* behavior).

> I'm leaning toward the view menu right under Show Snap Grid and above Use Perspective Projection.

Yeah, would be good. So, in menu "View" it would look like:

- View
  - ...
  - [ ] Show Snap Grid <kbd>\></kbd>
  - [x] Fade previous solid groups
  - [ ] Use Perspective Projection <kbd>`</kbd>
  - ...

phkahler, 2020-12-13:
> In the next _SolveSpace_ releases there should be more flexible setting for this such as:
> 
>     * Fade Previous Solid Groups
>       
>       * `(●)` All Previous Groups
>       * `( )` Only Assembly Groups
>       * `( )` (TBA; more features later)
>       * `( )` Do Not Fade

That's why I asked if it should be per-group. We already have a checkbox to not show the solid model for each group. For now I just want to help your dimensioning methods by not shadowing the solids.

Symbian9, 2020-12-13:
> We already have a checkbox to not show the solid model for each group.

Can you show me a screenshot of what you mean under *"a checkbox to not show the solid model for each group"*?

phkahler, 2020-12-13:
[image]

Right there under the section heading "solid model as" the is a checkbox "suppress this group's solid model". If that isn't what you're looking for, it is an example of where we might put group-specific visibility and rendering options.

Symbian9, 2020-12-13:
>  "suppress this group's solid model"

Enabling this option I just will fully disable some boolean operation.

This is NOT what I mean talking about about shadowing previous groups & it would not provide expected result!

> For now I just want to help your dimensioning methods by not shadowing the solids.

Actually if add *"Sketch in 3D"* or *"Sketch in Plane"* group as latest group of the 3D model, just after some *solid groups* (e.g. any group that brings boolean operation), built-in style `s00d-#def-dim-solid` **overrides all styles & colors of each previous solid groups** (temporary, until next solid group would be created).

So I requests here an option *"View >  Fade Previous Solid Groups"* (which by default should be enabled).

And if user would disable/uncheck *"View >  Fade Previous Solid Groups"* — all solids should be shown in its own style.

phkahler, 2020-12-20:
On a related tangnet:  I've been looking at code for drawing group and I think the variable "showShaded" is always true. If so there are a few redundant conditions. There are other flags as well, like "showFaces" and "drawOccludedAs". but I can't find anywhere that showShaded is set to false. I even put in a debug print to tell me if it was ever false and nothing ever printed.  Either I missed something or this is an old piece of cruft that never got removed.

phkahler, 2020-12-20:
Looks like the relevant code is in groupmesh.cpp line 572:
```
            if(type == Type::DRAWING_3D || type == Type::DRAWING_WORKPLANE) {
                fillFront.color = Style::Color(Style::DIM_SOLID);
            }
```
Just need to figure out how to add a checkbox item in the menu and add that condition here.

I still don't understand how "showShaded" gets set to false by the toolbar icon but that's not important.

Symbian9, 2020-12-21:
> I still don't understand how "showShaded" gets set to false by the toolbar icon but that's not important.

If I think in right direcrtion, togling from [image] to [image] just set all faces colors to main window background color + set ambient to `1.0`, see my comment:

- https://github.com/solvespace/solvespace/issues/524#issuecomment-628433539 

Or you mean something else?

ruevs, 2020-12-22:
> I still don't understand how "showShaded" gets set to false by the toolbar icon but that's not important.

https://github.com/solvespace/solvespace/blob/14e837a45f599b9b2a8a28d8628f96490b593c24/src/textwin.cpp#L170

https://github.com/solvespace/solvespace/blob/14e837a45f599b9b2a8a28d8628f96490b593c24/src/textwin.cpp#L69

;-)

closed 2020-12-29 by phkahler