solvespace/solvespace #835: OSX 3.0RC1 Export 2D view issue with "export these objects" line style option

chamnit, 2020-12-01:
### System information

SolveSpace version: v3.0 3.0~bcb8cd2c

Operating system: OSX Catalina 10.15.7

### Expected behavior

"Export 2d view..." of sketch should have included all sketched lines or any lines that are designated with the user line style with the 'export these objects' option checked and enabled. Sketch composed of lines that are not closed contours. Trying to draw cut lines for a laser cutter to follow. 

### Actual behavior

Exports only the 'closed contour' part of the sketch. Seems to ignore the non-closed contour lines. Moving non-closed contour lines to a user-defined line style with 'export these objects' enable also has no effect. Still missing in output file.

Tested with Windows version (same release, Win7) and the test SLVS file exported as expected with these non-closed contour lines in the PDF output.

### Additional information

Simple sketch of a trapezoid with an additional line through the middle. The middle line does not appear in the exported 2d view file on the OSX version of SolveSpace.

[test.slvs.zip](https://github.com/solvespace/solvespace/files/5625194/test.slvs.zip)

ruevs, 2020-12-01:
Indeed works fine on Windows.
[untitled.pdf](https://github.com/solvespace/solvespace/files/5625490/untitled.pdf)

phkahler, 2020-12-02:
Related or same as #682

chamnit, 2020-12-02:
@phkahler : The issue is related, but not the same. This example file I uploaded does not use the "export these objects" option. It's a standard sketch of a trapezoid and an additional line going through it, all in the same active group. The 'export 2d view' tool does not draw this additional line. 

If I try to move this additional line to a line-style with the "export these objects" option enable, there is no change. The additional line stills fails to be drawn in the export.

I updated the issue title to denote there is a problem with both.

chamnit, 2020-12-02:
For whatever reason, this test file works fine on my MacBook Air (same OS X and SolveSpace versions) but not on my Mac Mini. I will investigate further to see if there are any obvious differences between the configurations on the two machines.

chamnit, 2020-12-02:
Didn't find much different, except the renderer. The MacBook Air was running the Intel HD Graphics 5000 OpenGL engine. The Mac Mini wasn't (Intel UHD Graphics 630). The latter is the one that doesn't work properly. 

Here are the PDF outputs from both. As you can see, the non-OpenGL version on the Mac Mini is missing the middle line, and the constraints are a bit messed up as well.

[Untitled_Air_OpenGL.pdf](https://github.com/solvespace/solvespace/files/5630880/Untitled_Air_OpenGL.pdf)
[Untitled_Mini_UHD_Graphics_630.pdf](https://github.com/solvespace/solvespace/files/5630881/Untitled_Mini_UHD_Graphics_630.pdf)

I did also notice one more thing that might be useful. The SolveSpace configuration option "export curves as piecewise linear" is stated just this way on the OpenGL version (Air). But on the non-OpenGL version (Mini), it states "export curves as piecewise linear (since cutter radius is not zero)". Not sure why this additional "(since cutter radius is not zero)" is there, but thought I'd mention it.

chamnit, 2020-12-02:
For another data point, SolveSpace on an Apple Silicon M1 MacBook Air runs fine (emulated via Rosetta). It exports the 2D drawing in the test file correctly and as expected.

ruevs, 2020-12-02:
You probably have the "cutter radius offset" option set to a non-zero value on the Mac Mini. Another question is why it causes the problem... I'm not on a computer right now but I'll check it later.
[image]

chamnit, 2020-12-02:
@rlutz : The cutter radius offset was set and shown as '0.00' but I was messing with with it not too long ago. I manually re-entered '0.0' on cutter radius offset (and '1.00' for the export scale factor since I was messing with that as well). The "(since cutter radius is not zero)" message disappeared.

Only after a restart of SolveSpace, the problem went away. The middle line was present and the constraints were drawn correctly. 

I suppose the issue is now solved. Makes sense that any non-enclosed contours in the sketch and constraints would be ignored by the cutter radius offset algorithm. 

However, it does seem weird that the cutter radius offset setting was shown as '0.00' and caused this issue. To help avoid this issue later, is there a more obvious way to shown if the cutter radius offset is active other than a 0.00 setting? 

I suspect that some numerical round-off happened at some point and inserted a very small number. Enough to enable it, but not denote that it was not zero.

phkahler, 2020-12-06:
There is a configuration option for "digits after decimal point to show".  Much as I'd like to think this has been resolved ( I suppose @chamnit is good now ),  the fact remains that a small non-zero value in there will change the way something works while remaining hidden.

Looking further, setting the digits to 3 and typing a value of 0.0004 the display show 0.000 as expected for rounding down and will show 0.0004 if the number of digits is increased to 4. However, with digits shown set to 3 and I try to edit the value it displays 0.000 in the edit box. If I press return to enter that value it is actually changed to 0, but if I click outside the edit box to make it go away the value remains 0.0004 which can only be seen by increasing the number of digits shown. In other words, even doing an edit to verify the value is zero can fail unless the user actually enters zero.

Is this a high priority? No.  But I think it should stay open with a title change, or closed with a new issue opened focusing on number of digits and/or the invisible behaviour change.

rlutz, 2020-12-06:
@chamnit Did you mean to mention me? I'm not sure how I'm related to this issue.

ruevs, 2020-12-06:
And here is the other open issue on the subject of significant digits in the text window:
https://github.com/solvespace/solvespace/issues/619

It is definitely a problem.

closed 2021-01-14 by phkahler