solvespace/solvespace #839: Short links to GitHub repo

Symbian9, 2020-12-08:
### Actual behavior

Just reserved short links via `git.io` to SolveSpace main repo on GitHub

- https://git.io/solvespace (redirect to `https://github.com/solvespace/solvespace`)
- https://git.io/SolveSpace (redirect to `https://github.com/solvespace/solvespace#solvespace`)
- https://git.io/SLVS (redirect to `https://github.com/solvespace/solvespace.git`; but in most cases it will redirect to `https://github.com/solvespace/solvespace`)

JFTR, ~~`git.io/slvs`~~ was reserved already by someone else.

### Additional information

- `git.io` Custom URL - https://gist.github.com/dikiaap/01f5e2ba3c738012aef0a8f524a6e207

vespakoen, 2020-12-08:
Thanks, although i'm not really sure what the use-case is for these.
The first link doesn't seem to work by the way.

Closing since it's not really an issue

closed 2020-12-08 by vespakoen

Symbian9, 2020-12-08:
> The first link doesn't seem to work by the way.

Oh, don't know what has happen with it.

BTW, think `git.io/SLVS` would just enough ;)