solvespace/solvespace #840: Incorrect capitalisation in CMakeLists.txt prevents OpenMP build

mderouss, 2020-12-08:
### System information

SolveSpace version: 3.0~b316a886

Operating system: XUbuntu 20.04

### Expected behavior

Setting ENABLE_OPENMP to ON should result in a build which does not throw unknown pragma warnings when it encounters #pragma omp . 

### Actual behavior

Several warnings complain that #pragma omp is not defined. 

### Additional information

The problem is that CMakeLists.txt tests 'OPENMP_FOUND' when deciding whether to set the OpenMP_CXX_FLAGS. It should test 'OpenMP_FOUND'. When this change is made, compilation proceeds as expected and no pragma undefined warnings are generated.

ruevs, 2020-12-09:
Hmm I am almost a complete novice with CMake, but it seems this should not be a problem:
https://github.com/apache/incubator-mxnet/issues/17579
https://gitlab.kitware.com/cmake/cmake/blob/master/Modules/FindOpenMP.cmake#L590

What version of CMmake are you using?
Does your `FindOpenMP.cmake` contain the line
``` 
set(OPENMP_FOUND ${OpenMP_FOUND})
```
?

(On Windows it is located in ...\CMake\share\cmake-3.19\Modules)

mderouss, 2020-12-10:
Ahem.  

Yes, I can see that line too in FindOpenMP.cmake. So I've just blown away my solvespace environment and started again from scratch ( 3.0~f5086b62 ). I built first with everything as it came from the repo, all is good, no pragma complaints. 

But this time, before I flipped the switch in CMakeLists.txt to turn OpenMP 'ON' and reran cmake, I blew away the entire contents of the build directory. Now everything is fine, i.e. no pragma complaints, and cmake has told me 

-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- found OpenMP, compiling with flags: -fopenmp

So apologies, looks like there is no mystery/bug, I just hadn't cleaned up adequately between switching builds :(. Closing.

closed 2020-12-10 by mderouss