solvespace/solvespace #841: Couple of issues drawing a symmetric drawing with a lot of tangent arcs

vespakoen, 2020-12-08:
### System information

SolveSpace version: 3.0 / edge (Release build with OpenMP)

### The issue

Just wanted to drop this test file over here, I am trying to draw this 8040 aluminium profile, like this one:

[image]

And the experience is not optimal, I think it might have to do with having a lot of tangent arcs, in any case, it is becoming pretty slow to draw.

Another reason why I am dropping this file here is because wanting to draw this made me find a couple of problems / missing features in SolveSpace.

This attached sketch only contains a quarter of the final drawing, and mirroring / rotating it was not possible because of #833 and (the now merged) #830, but even with those changes merged, it complaints that there are `Too many items to paste`.
I think this should be a higher number by default, I doubled it and the actual pasting of the drawing wasn't slow at all (maybe it's memory intensive?).

Even with the possibility of rotating and mirroring the drawing, It also duplicates all the dimension constraints 4 times, and of course in this drawing it would make more sense to control all the dimensions in only a single place.

So I guess what makes more sense here is to make a new group by step rotating / translating, however, this does not allow us to mirror yet, so it's not a really good option.

I have experimented with drawing the complete "top half" of the drawing and step rotating it, which works, but then I run into the issue that I cannot extrude the resulting drawing when imported in another sketch.

So in conclusion:

1) The drawing is slow to solve, this might actually already be fixed with other PR's, but I guess this file can be handy as a test case.
2) When using `Paste Transformed`, I think there would be a lot of added value if there would be an option to change dimension constraints into equal length constraints.
3) There is probably value in having a "Symmetric" group tool that can copy and mirror the drawing over 1 or 2 axes.
4) There is an issue extruding a Step Rotated (haven't tested Step Translated drawings) drawing when imported into another sketch.

[8040.slvs.zip](https://github.com/solvespace/solvespace/files/5658434/8040.slvs.zip)

phkahler, 2020-12-08:
Did you try the 2020 or 4040 I included in #793 zip file?  I made a construction sketch with all the critical dimensions and then another with just the entities you'd want when linking them. I didnt do it for performance, but to cut down on extraneous stuff when linking. The first group entities are hidden.

I'm not certain, but I think the @Evil-Spirit patches (non Eigen) may help performance when linking sketches with a lot of entities - like this or an IDF file. Linked sketches only have 6 DoF but may have hundreds of constraints, all of which might be solvable by substitution which he made faster.

vespakoen, 2020-12-08:
That is an interesting technique for hiding extraneous things, thanks for that!

I would could probably dumb down my profile drawing as well, but I was interested in providing this as a test case, that hopefully allows us to speed up the solving a bit more.

6) I guess adding some benchmarks for the solver would also be helpful to see what the impact is of certain PR's.

I was able to rotate and flip the drawing a couple of times using Evil-Spirit's changes (including Eigen).

The linking / extruding is all working fast, it's just the (complex) drawing itself that is (very) slow to work on.

7) By the way, when "Too many unknowns" is "active", It seems to be impossible to add constraints (maybe it's possible when relaxing stuff?) and also the undo history gets spammed when moving stuff around when it's "active".

Should probably split all of this up into separate issues, but maybe they can be discussed here first.

phkahler, 2020-12-09:
Benchmarks are good. I've used a lot of ad-hoc tests when making performance improvements.  Michael over at Phoronix asked if there's anything useful in our "bench" folder, but sadly I had to say I have no idea. Making something useful and automated could maybe get us in his test suite ;-)

I have no doubt Evil's patches help, I'm just curious how much is from his other changes vs Eigen. They probably affect different use cases.

7) what is "Too many unknows"?  I am also still learning SolveSpace.

phkahler, 2020-12-09:
>     5\. There is an issue extruding a Step Rotated (haven't tested Step Translated drawings) drawing when imported into another sketch.

Did you try extruding the original (unlinked) rotated sketch? Your unrotated version should not be a closed sketch. Allow the rotated copies to meet at their ends to close it.

vespakoen, 2020-12-09:
Our bench folder contains a little script that can load up a .slvs file and will tell you how long it took to "generate".
I guess it is a start, it would be nice to be able to compare results from it with past builds. something along the lines of: `This PR slows down SolveSpace 2.4x times compared to the master branch` in a check result.

I have tried applying Evil-Spirit IdList indexing patches, however, they seem to be so old and a lot of it seems to be implemented differently now, so I think those are out.

I will try to see what the differences are with and without Eigen.

I tried extruding the original sketch, that worked, but when linked into another file it didn't work anymore, I'll try this out again and make a new issue for it if I can re-create it.

ruevs, 2020-12-09:
>     5. There is an issue extruding a Step Rotated (haven't tested Step Translated drawings) drawing when imported into another sketch.

This actually works exactly as intended. See here https://github.com/solvespace/solvespace/issues/838#issuecomment-742104563

ruevs, 2020-12-09:
@vespakoen Your 8040 was two rotates and two `G`s (Toggle Construction) away from being perfectly fine (sans the center hole).

Here it is [8040_finished.zip](https://github.com/solvespace/solvespace/files/5669215/8040_finished.zip)

[image]

Constructed like this it is not too slow.

ruevs, 2020-12-10:
And a bit more finished (I could not help myself).
[8040_finished_more.zip](https://github.com/solvespace/solvespace/files/5669397/8040_finished_more.zip)
The central opening is not fully constrained since I do not know one radius.
By the way how do you get the 1.8mm wall thickness? It comes out of the constraints but I could not parse it :-)

And the solving when editing the sketch is indeed slow. I did not notice it before because I just rotated it a couple of times.

[image]

ruevs, 2020-12-10:
Aha! I figured it out. Here it is fully constrained.

[8040_finished_constrained.zip](https://github.com/solvespace/solvespace/files/5669510/8040_finished_constrained.zip)

By the way the release build is much, much faster when solving. Which is not an excuse to not improve it ;-)

phkahler, 2020-12-10:
This becomes almost unusable for me. I can make one extrusion, but doing step-translating after that kills it. It spends a lot of time running on only one core.

vespakoen, 2020-12-10:
Sexy!! That double rotating makes perfect sense now! and also thanks for explaining why the linked extrude wasn't working for me and for finishing the drawing! =)

It is a really slow drawing to solve indeed, and I will be using this file as a test when I get some kind of decent profiler up and running.

I imported a .dwg file for this profile, "measured it" using solvespace then drew it again.
Feel free to use it anyway you like!

ruevs, 2020-12-10:
To use the file for linking (importing) and extruding in other assemblies the file needs to be prepared like this:
[8040_ForImport.zip](https://github.com/solvespace/solvespace/files/5673003/8040_ForImport.zip)

Since only `g004-rotate` forms a closed contour in needs to be the only one enabled. Otherwise when importing and trying to extrude the non-closed contours imported from (enabled) `g002` and `g003` would result in a non solid extrusion.

vespakoen, 2020-12-10:
Ahhhh now it clicked =) thanks so much for taking the time to figure this out! Opens up a whole new way of modeling in SolveSpace for me!

ruevs, 2020-12-10:
No problem :-)

So to sum up, following your points above:

1. The Solver is slow with long chains of constraints:
This has been discussed before. Possible solution - someone 
needs to understand, adjust, fix, amend and integrate the @Evil-Spirit patches (I want to try, but never find time).
References:
https://www.youtube.com/watch?v=wqrefKghpQE
https://github.com/solvespace/solvespace/issues/225, https://github.com/solvespace/solvespace/pull/432, https://github.com/solvespace/solvespace/pull/717, https://github.com/solvespace/solvespace/pull/721, https://github.com/solvespace/solvespace/pull/842, https://github.com/solvespace/solvespace/issues/131 (https://github.com/Evil-Spirit/solvespace-master/commits/master-131), https://github.com/solvespace/solvespace/issues/386 
2. Paste transformed changing constraints:
In my opinion this would just be confusing. A lot of (creative) Copy-Pasting (transformed or not) usually indicates that there is a better way ;-)
3. Copy-Paste a lot:
In my opinion not necessary. See #2 above.
4. Symmetric group:
Completely covered by rotating - by picking an appropriate axis to rotate around. But also see https://github.com/solvespace/solvespace/pull/592, https://github.com/solvespace/solvespace/issues/614, https://github.com/solvespace/solvespace/issues/590
5. Extruding linked (imported) sketches:
Works as intended - see above.

BTW for anyone reading in the future the [reference](https://solvespace.com/ref.pl) and [tutorials (including the videos)](https://solvespace.com/tutorial.pl) are really good and worth studying.

vespakoen, 2020-12-10:
1) Thanks for all the references! I will play around with those
2,3,4) Agreed, again thanks for the references!
5) Thanks

I'm learning =)

closed 2020-12-10 by vespakoen

ruevs, 2021-04-07:
@vespakoen @phkahler Hey! With this https://github.com/solvespace/solvespace/pull/1002 the 8040 model is wicked fast! No slow down at all while editing the constraints!