solvespace/solvespace #845: Enhancement: allow copy and paste between different instances of Solvespace

ardugitter, 2020-12-09:
### System information

SolveSpace version: 3.0~bcb8cd2c

Operating system: Debian stable

### Expected behavior

**This is a feature request: please could it be made possible to copy and paste between multiple instances of Solvespace?**

ruevs, 2020-12-09:
This would be useful and... complicated... especially on a cross platform application.

Take a look at  (if you have not discovered it yet) "New Group | Link/ Assemble..." `Shift+I` in case it helps you. It is described in the [tutorials.](https://solvespace.com/tutorial.pl)

ardugitter, 2020-12-09:
I agree that it is potentially complicated, BUT, this could be simply implemented by allowing the action of Ctrl-C to save the copied data to disk in a temporary file in the current working directory. e.g. filename ".solvespace_copy"

If you used Ctrl-X, you would have filename ".solvespace_cut"

Then, any other instance of solvespace in the same directory could pick up on the presence of this file when Ctrl-V was used. Of course, in the case of the file detected being called ".solvespace_cut", you would delete the file after pasting it into the current Solvespace instance.

In this way, a cut, copy and paste could easily be implemented and shared between multiple Solvespace instances.

baryluk, 2022-12-08:
I just had a need to use this. If this cannot be done nicely in cross platform way, it would be great if a selection can be saved to disk, and then pasted from file.

Yes, I know of Link/Assemble, but sometimes it is not easy. Also this requires planning. What if I want to make a new group, from an existing file, or copy a bit of one sketch to another, and change it (they should not share anything with each other) for experimenting with various minor variations.