solvespace/solvespace #847: Space Navigator gone?

phkahler, 2020-12-10:
There are people on the forum noticing that Spacemouse is not working. i looked at the build instructions and under Windows it says it will no be supported.  I don't have one, but when was this removed? Is it still supported on Linux or macOS?

jwesthues, 2020-12-11:
I don't think it's actually removed? The code still seems to be there, but `#ifdef HAVE_SPACEWARE`. So I'm thinking the standard build just doesn't enable that, perhaps because the library in question is an opaque binary blob from the hardware vendor, rather impure. I'd have no objection to re-enabling, though.

phkahler, 2020-12-12:
Do we statically link the blob or is there a header file for it or what? Also, how does this vary by platform?

jwesthues, 2020-12-12:
Under Windows, statically linked, library and headers at

https://github.com/solvespace/solvespace/tree/master/extlib/si

This library is provided by the hardware vendor. It's a 32-bit library, so it's disabled for 64-bit builds, which perhaps is the reason for the statement in the build instructions you note above. That doesn't explain why it would be broken in the RC though. Perhaps check if the build process is actually finding the libraries and enabling `HAVE_SPACEWARE`?

I didn't write the support for other platforms, so I'm less familiar there. On brief inspection, looks like Mac uses the vendor-provided library, but dynamically links to a version that's already installed. Linux uses libspnav, an independent and free implementation.

phkahler, 2020-12-12:
@vespakoen and @ppd  I looked at a recent CI build from one of the PRs and it does print the line "Using prebuilt SpaceWare" for the Windows build.  From the CMakeLists.txt that would be here:

```
        if(MSVC AND ${CMAKE_SIZEOF_VOID_P} EQUAL 4)
            message(STATUS "Using prebuilt SpaceWare")
            set(SPACEWARE_FOUND TRUE)
            set(SPACEWARE_INCLUDE_DIR
                "${CMAKE_SOURCE_DIR}/extlib/si")
            set(SPACEWARE_LIBRARIES
                "${CMAKE_SOURCE_DIR}/extlib/si/siapp.lib")
        endif()
```
However I don't see any other references to SPACEWARE.  How to tell if it's actually getting linked?

macOS seems to always build with it, while for guigtk.cpp there is this nice condition:
```
#if defined(HAVE_SPACEWARE) && defined(GDK_WINDOWING_X11)
```
That GDK_WINDOWING_X11 has me scratching my head.

ruevs, 2020-12-14:
It gets enabled by this https://github.com/solvespace/solvespace/blob/cac3aae0c7d43413897490859b7d2518efd64c31/src/CMakeLists.txt#L9, which sets `HAVE_SPACEWARE` based on `SPACEWARE_FOUND`.

It gets defined for me on Windows and the debugger breaks here https://github.com/solvespace/solvespace/blob/cac3aae0c7d43413897490859b7d2518efd64c31/src/platform/guiwin.cpp#L1399 so it should work.

I don't have a SpaceMouse to test but am attaching the current master compiled on my machine for someone who has a SpaceMouse to test.
[solvespace_win32_cac3aae0c7d.zip](https://github.com/solvespace/solvespace/files/5689354/solvespace_win32_cac3aae0c7d.zip)

atomicdog, 2020-12-14:
I tried out your win32_cac3aae0c7d build but SpaceMouse is still not working for me.

kk6mrp, 2020-12-15:
I just got my Space Mouse back from my cousin and it does work on Linux with X11 but I haven't figured out how make it work under Wayland yet.

I will test it out under Windows later.

phkahler, 2020-12-15:
@kk6mrp  I'm not sure it ever worked on Wayland. The GDK_WINDOWING_X11 condition I highlighted above may suggest it only works on X but I'm not sure.  Windows was where the problem was reported.

@ruevs That's strange. The code you stopped on is conditionally compiled, so it is there but doesn't work?

kk6mrp, 2020-12-15:
@phkahler I can't find any documentation on spacenavd/libspnav even working with Wayland so I'm guessing it has never worked.

ruevs, 2020-12-15:
https://github.com/solvespace/solvespace/pull/532 Just for reference.

kk6mrp, 2020-12-17:
@ruevs I finally figured out how to get MSVC to compile and I started to do a bisect for this. I could build the latest release but am having issues building earlier releases. Right now I'm at https://github.com/solvespace/solvespace/commit/1e2f1996335120c9d707d1b15f2fe2f56521d30b and it will build solvespace but the executable will not run. Is there something I'm missing here?

[solvespace.zip](https://drive.google.com/file/d/1gx4iBtDtGxprnQ__yMkg3Aof5glm6Fwv/view?usp=sharing)

ruevs, 2020-12-17:
Yeah, some revisions around this are broken. Just do `git bisect skip` on them.

kk6mrp, 2020-12-17:
Ah.. Ok, I'm not real familiar with the Windows workflow yet. Thank you!

ruevs, 2020-12-21:
@kk6mrp could you figure out when it stops working?

kk6mrp, 2020-12-21:
Yes, I did the other day, it stops at your bug fix https://github.com/solvespace/solvespace/commit/8ef3cacc332c004a364e1e65c308e2892058905f.
There is another regression that I haven't been able to narrow down yet also.

ruevs, 2020-12-22:
Interesting! So it works **_before_** 8ef3cac and does not work after it?!
This is truly weird!

The change in my commit seems like it should make it work after https://github.com/solvespace/solvespace/blame/f6a774d7bf4896716219bcd846ddd4341c4abaa2/src/platform/guiwin.cpp#L1392 c1f1c7c40 which appears to have broken it. And not the other way around. And even @whitequark who authored the commit approved my fix.

Can you please place break points at the three places where you find `if(hSpaceWareDriverClass != NULL)` and debug it?

kk6mrp, 2020-12-22:
The only difference I see is it being declaring as static after the platform abstraction. In the 3Dconnexion forums I couldn't find any reference or examples that it should be declared that way, but I'm not familiar enough with C++ to know which way it should be.

I'll let you know what I find when I get a chance to work on it hopefully tomorrow evening.

ruevs, 2020-12-23:
> The only difference I see is it being declaring as static after the platform abstraction.

If you mean this
https://github.com/solvespace/solvespace/blob/14e837a45f599b9b2a8a28d8628f96490b593c24/src/platform/guiwin.cpp#L1396
ten the `static` certainly makes no difference.

robnee, 2020-12-30:
I notice this problem too.  I tried searching for the window class "SpaceWare Driver Class" using a short AutoHotKey script and it did not find it.  That would the code to skip the init of the SpaceMouse support in SolveSpace. Could there have been a change to the Window class name?  There was recently (dec 20, 2020) a driver update for the SpaceMouse that might have changed something?

Rob

kk6mrp, 2020-12-30:
It is using the same window class in Solvespace 2.3 as now so the problem is somewhere else in Solvespace. It doesn't work with the older Windows 7 drivers either.

@ruevs I'm going to have to try out the bisect again. I tested tried testing out commits before https://github.com/solvespace/solvespace/commit/8ef3cacc332c004a364e1e65c308e2892058905f and the SpaceMouse wasn't working either. So I must have done something wrong with the Bisect.

phkahler, 2021-01-15:
Has anyone made progress on this issue? It would be a shame not to have a fix for V3RC2.

jwesthues, 2021-01-16:
I dusted off my old SpaceMouse Compact, and it should be fixed in 76589a8a879218f7ed67ffbbcd8070086baa6fe8. We were creating the event but never passing it to `window->onSixDofEvent()`, so I suspect this has been broken since the refactor? But I'm not sure why it would work before 8ef3cacc332c004a364e1e65c308e2892058905f (and I haven't gone bisecting myself), so please review and confirm.

kk6mrp, 2021-01-16:
Your commit https://github.com/solvespace/solvespace/commit/76589a8a879218f7ed67ffbbcd8070086baa6fe8 does indeed fix the issue! There were so many builds failing that I think I must have confused myself when doing the bisect. The regression started much sooner than https://github.com/solvespace/solvespace/commit/8ef3cacc332c004a364e1e65c308e2892058905f.

phkahler, 2021-01-16:
@kk6mrp thank you for the confirmation. I'll close this now.

phkahler, 2021-01-16:
Fixed in https://github.com/solvespace/solvespace/commit/76589a8a879218f7ed67ffbbcd8070086baa6fe8

closed 2021-01-16 by phkahler