solvespace/solvespace #849: Linked step rotating sketch cannot be extruded

justthinkering, 2020-12-10:
[extrude-linked-sketch.zip](https://github.com/solvespace/solvespace/files/5675977/extrude-linked-sketch.zip)
### System information

SolveSpace version: 3.0~b316a886

Operating system: Arch Linux

### Expected behavior

Create a sketch using the step rotating tool (shift + r), link it into another file (shift + i) and then create a solid via extruding (shift + x).

### Actual behavior

* No solid model is created.
* The extrusion shows only the edges of a 3D object. This looks similar to the extrusion of a sketch not having a closed contour, but without linking, the step rotating sketch can be extruded as expected.
* After linking, the step rotating sketch does not show the typical coloring of the enclosed surface anymore.

### Additional information

ruevs, 2020-12-10:
It works as intended see here https://github.com/solvespace/solvespace/issues/838#issuecomment-742104563, https://github.com/solvespace/solvespace/issues/838#issuecomment-742519412, https://github.com/solvespace/solvespace/issues/841#issuecomment-742518861

closed 2020-12-10 by ruevs

justthinkering, 2020-12-11:
Thank you very much!