solvespace/solvespace #852: compiling on apple m1

aep, 2020-12-12:
need a little help here to get it compiling on apple m1 since i lack the cmakefoo

i commented out everything with libomp in CMakeFile.txt 

```diff
@ if(ENABLE_LTO)
 endif()

 if(ENABLE_OPENMP)
-    find_package( OpenMP REQUIRED )
-    if(OPENMP_FOUND)
-        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
-        message(STATUS "found OpenMP, compiling with flags: " ${OpenMP_CXX_FLAGS} )
-    endif()
```

as a starting point, but its still trying to link a hardoded path.

```
cp: /usr/local/opt/libomp/lib/libomp.dylib: No such file or directory
```

where is that path hardcoded? somewhere in cmake itself?

aep, 2020-12-12:
found it in another CmakeFile,  next issue looks harder

```
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_ASM_COMPILE_OBJECT
```

phkahler, 2020-12-12:
@aep See pull request #851.

BTW I assume you've tried the Intel version on M1, how well does that work?

closed 2020-12-12 by aep