solvespace/solvespace #853: Prep for GTK4

phkahler, 2020-12-12:
DON'T PANIC!!
This is only preparation for a future change and would only be the first of the two [lists here](https://developer.gnome.org/gtk4/stable/gtk-migrating-3-to-4.html) which can be done while still using GTK3. 

There is no hurry to use GTK4 but some day it will be necessary and we might as well know our status. It shouldn't be hard since we don't use many platform UI features. The list is here so we can check them off when they are done or confirmed as non-issues for SolveSpace - there is more detail at the link above.

 * [ ] Do not use deprecated symbols
 * [ ] Enable diagnostic warnings
 * [ ] Do not use widget style properties
 * [ ] Review your window creation flags
 * [ ] Stop using direct access to GdkEvent structs
 * [ ] Stop using gdk_pointer_warp()
 * [ ] Stop using non-RGBA visuals
 * [ ] Stop using GtkBox padding, fill and expand child properties
 * [ ] Stop using the state argument of GtkStyleContext getters
 * [ ] Stop using gdk_pixbuf_get_from_window() and gdk_cairo_set_source_surface()
 * [ ] Stop using GtkWidget event signals
 * [ ] Set a proper application ID
 * [ ] Stop using gtk_main() and related APIs
 * [ ] Reduce the use of gtk_widget_destroy()
 * [ ] Reduce the use of generic container APIs</title>

Symbian9, 2021-01-11:
> DON'T PANIC!!

JFTR, What about make SolveSpace fully independ from GTK and make just 'single-window' OpenGL canvas? (that would also solves "Single Window" https://github.com/solvespace/solvespace/issues/39)

phkahler, 2021-01-11:
> JFTR, What about make SolveSpace fully independent from GTK and make just 'single-window' OpenGL canvas? (that would also solves "Single Window" #39)

Not really an option. We support 3 GUI toolkits (Windows macOS and GTK) but all we use them for is creating a window, menus, popup dialogs, mouse/keyboard events, and.... I think that's it.  The entire text window, toolbars, and 3D view are painted using OpenGL already. There is no such thing as creating a window with OpenGL, so something has to be used for that. If we used SDL that should work on all platforms, but then we'd have to draw our own menus and still need to use something for file selector and dialogs. Changing GUI toolkits is not going to happen. Someone is free to implement an additional GUI, and if that works on all platforms well enough we could deprecate the old code - but I say that in jest, nobody is going to come along and do that, not even with Qt so far. Let me rephrase that - not even KDE users care enough to make a Qt version.

Single window could be done using each GUI toolkit to create another pane within the main window and drawing the text window there. IIRC whitequark indicated that emscripten wouldn't support 2 GL windows (not that I care about emscripten). Also, we'd have to do it on all 3 platforms. The other option is to draw everything - including any separator - in the main window. I have not looked at the single window branch, so I'm not sure what's in there.

I would prefer to dump all the other GUIs and just use GTK on all platforms, but that's just extra work and would lead to degradation of the platform abstraction layer. GTK is also not great outside of Linux AFAICT.  Migrating to GTK4 is also an excuse to become more familiar with the GUI code. We use so little of it that we might be able to build against GTK3 or 4 with the same code at some point.

Symbian9, 2021-01-11:
> ... all we use them for is creating a window, menus, popup dialogs,... 
>
> There is no such thing as creating a window with OpenGL, so something has to be used for that

My idea is to draw *windows, menus & popup dialogs* directly in main window same way as ***Toolbar*** ("`[v] Show Toolbar" ) drawn.

phkahler, 2021-01-14:
When I enable diagnotic warnings at runtime the open-file dialog consistently spits out this error:
```
(solvespace:458388): Gtk-CRITICAL **: 20:51:28.383: gtk_file_chooser_widget_set_current_name: assertion 'priv->action == GTK_FILE_CHOOSER_ACTION_SAVE || priv->action == GTK_FILE_CHOOSER_ACTION_CREATE_FOLDER' failed
```
Seems like GTK_FILE_CHOOSER_ACTION_SAVE  or  GTK_FILE_CHOOSER_ACTION_CREATE_FOLDER would be for a save-file, not an open-file.  There is no error shown when doing File->SaveAs.

BTW, anyone can do this with current release builds by starting with:
G_ENABLE_DIAGNOSTIC=1 bin/solvespace     or whatever path you use.

vthriller, 2023-08-31:
> I would prefer to dump all the other GUIs and just use GTK on all platforms [...] GTK is also not great outside of Linux AFAICT.

What do you think of Qt?

ruevs, 2023-08-31:
> What do you think of Qt?

#401 

And from https://github.com/solvespace/solvespace/issues/1024 :

## Alternative UI toolkit/framework?
- [Dear ImGui](https://github.com/ocornut/imgui) - https://github.com/solvespace/solvespace/issues/263
- [Qt](https://www.qt.io/) - https://github.com/solvespace/solvespace/issues/401
- [FLTK](https://www.fltk.org/) - https://github.com/solvespace/solvespace/issues/1117
A FLTK port was added on 2013-10-28 in c60e3dd34 and removed (in favor of GTK3 and Cocoa) on 2015-07-10 in d0c0a7ae5a9181d66b1eb272f083ba28f0ef2887.
- GTK 4 - https://github.com/solvespace/solvespace/issues/853

phkahler, 2023-08-31:
> What do you think of Qt?

Aside from being a useful toolkit, I think it's big and bloated and in a constant state of tension between open source and commercial. But if someone wants to write platform/guiqt.cpp so we're native on KDE I'd probably merge it. But we don't really use enough GUI at this time to make a difference.

phkahler, 2023-09-29:
So I quit looking for specific things in the source and just grepped for gtk_  and found there is only one instance which is gtk_show_uri_on_window() which I recall dealing with recently and I think is gtk4 approved. Now searching for gdk_  gets about 100 hits.

This makes sense, as GDK is an abstraction that GTK uses to remain OS agnostic and we barely use any actual GTK stuff.

phkahler, 2023-10-01:
@rpavlik  @ppd  @vespakoen 

Where in the build can I pass -std=C++17 or -std=gnu++17  to the compiler? Turns out using gtk4 wants a newish compiler.

phkahler, 2023-10-01:
More specifically I get:
```
Building CXX object src/CMakeFiles/solvespace.dir/platform/guigtk4.cpp.o
In file included from /usr/include/sigc++-3.0/sigc++/functors/slot.h:23,
                 from /usr/include/sigc++-3.0/sigc++/signal_base.h:27,
                 from /usr/include/sigc++-3.0/sigc++/signal.h:25,
                 from /usr/include/sigc++-3.0/sigc++/sigc++.h:118,
                 from /usr/include/glibmm-2.68/glibmm/signalproxy.h:27,
                 from /usr/include/glibmm-2.68/glibmm/objectbase.h:22,
                 from /usr/include/glibmm-2.68/glibmm/object.h:29,
                 from /usr/include/glibmm-2.68/glibmm/binding.h:22,
                 from /usr/include/glibmm-2.68/glibmm.h:103,
                 from /usr/include/gtkmm-4.0/gtkmm.h:100,
                 from /home/phkahler/projects/solvespace/src/platform/guigtk4.cpp:13:
/usr/include/sigc++-3.0/sigc++/visit_each.h:37:24: error: ‘decay_t’ is not a member of ‘std’; did you mean ‘decay’?
   37 |   std::is_base_of<std::decay_t<Base>, std::decay_t<Derived>>::value ||
      |                        ^~~~~~~
      |                        decay
/usr/include/sigc++-3.0/sigc++/visit_each.h:37:24: error: ‘decay_t’ is not a member of ‘std’; did you mean ‘decay’?
```

phkahler, 2023-10-03:
Had to bump to -std=c++17 which was not as simple as I thought. Now dealing with Gtk::CheckMenuItem not being a thing any more.

phkahler, 2023-10-03:
This seems very important:
https://stackoverflow.com/questions/73497163/how-to-programmatically-add-a-check-mark-to-a-gmenuitem-gtk4-using-c

phkahler, 2023-10-15:
So not only should we not directly reference event structure members, we should stop using events and use event controllers. Since the API docs only describe what but not how:
https://blog.gtk.org/2020/04/29/custom-widgets-in-gtk-4-input/

rpavlik, 2023-10-16:
> @rpavlik @ppd @vespakoen
> 
> Where in the build can I pass -std=C++17 or -std=gnu++17 to the compiler? Turns out using gtk4 wants a newish compiler.
> 
> Answer self: -DCMAKE_CXX_FLAGS=-std=gnu++17 added to CMAKE
> 
> Next up, I get errors in the includes when I try to build my GTK4 branch: https://github.com/phkahler/solvespace/tree/gtk4
> 
> BTW it isn't an option on that branch, it only tries to build GTK4 on Linux.

CMAKE_CXX_STANDARD 17
CMAKE_CXX_STANDARD_REQUIRED true

phkahler, 2023-10-17:
> CMAKE_CXX_STANDARD 17

Which is in the top level CMakeLists.txt here:
https://github.com/solvespace/solvespace/blob/e7c0c1665f1684bb3195107147aaf254c852fa44/CMakeLists.txt#L14