solvespace/solvespace #855: Show only SLVS file name in the title of main window (NOT full path)

Symbian9, 2020-12-14:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/2939abf5f876a7454e0afbc06b32c888989b6546
- **Operating system:** Debian 10 buster

### Expected behavior

Show only SLVS file name in the title of main window, eg. `<file_name>.slvs`

### Actual behavior

SolveSpace shows full path to SLVS file in the title of main window, eg. `/path_to/<file_name>.slvs`.

### Additional information

[image]

ruevs, 2020-12-14:
Why?!

Symbian9, 2020-12-14:
> Why?!

For the first, for security reasons! (title should not show user PC name, which is shown actually in path if file stored in home directory)

For the second, if there are few windows opened with different files but in same folder it would be much easier to select needed window on global/system window panel.

ruevs, 2020-12-14:
Having the full path is extremely useful. As a counter example to yours: if there are many files opened with the same name but in different directories it would be much easier to select...

 The cinvenience/"security" of "tutorial makers" should be completely disregarded. They should know what they are doing.
- Windows: Put the files for your tutorials in "C:\verysecret\..."
- Linux... I don't know, be root, make "/very secret" and give your very secret user name permissions, do the tutorial and delete it.
- Or remove your user name from the screen shot if you are not making a video.

Symbian9, 2020-12-14:
> Having the full path is extremely useful.

Having the full path is extremely useful *only for geeks*.

@phkahler, as compromise I propose add CLI option `--show-paths` to start SolveSpace with behavior similar to "old school" (aka SolveSpace `2.x`) behaviour.

ruevs, 2020-12-14:
Whether it is extremely useful only for geeks is hard to know, but it is possible.

On Windows even "common users" put their files all over the place (not only in their user directory). And enen the most recent Windows Explorer has the [“Display the full path in the title bar”](https://www.onmsft.com/how-to/how-to-make-windows-10s-file-explorer-show-the-full-directory-path-in-the-title-bar) option.

On Linux... how many Linux users are not "geeks"? ;-)

As for adding an option - in my opinion adding extra options (and thus code) for such things (fluff) is a path that eventually leads to software that I dont like.

There is another aspect that is underappreciated - change for tha sake of change has a price - more commits in the history that are noise, less useful "git blame" etc. So if a change has dubious benefit or it is not clear that it is an improvement, and nothing broken as it is I will argue against the change most of the time.

This is why I am arguing... but this is open source...

Symbian9, 2020-12-14:
> There is another aspect that is underappreciated - change for tha sake of change has a price - more commits in the history that are noise, less useful "git blame" etc.

Lets don't worry of commits number & just make SolveSpace looks more professional & modern for common users.

Just take a look on Blender, Krita, GIMP, LibreOffice, Inkscape, Scribus or at least on FreeCAD & QCAD.

Symbian9, 2020-12-14:
@ruevs @phkahler, please, reopen this issue as it is important SolveSpace UX issue which should be solved for `3.0` release.

ruevs, 2020-12-14:
OK I reopened the issue, but my comment above remains.

Better to make it "no full path" only than adding an option.

But all of this is just noise.

AVHon, 2021-01-18:
I prefer that programs show the full path in the title bar.

* It makes it easier to remember where my project is if I need to open another file
* If I have identically-named files in different places (archives, downloads, temporary directories, versions from different people, versions from different projects), it makes it easy to tell which one I'm looking at.

Symbian9, 2021-01-18:
> I prefer that programs show the full path in the title bar.

That is why I requested adding option into preferences, so user would choose show it or not ;)

dgramop, 2024-03-25:
fwiw, some window managers that truncate window titles (for example however they list windows) would make differently-named files that appear in the same /very/long/path/.../ all have the same truncated name

ruevs, 2024-06-11:
Merged here 71ad5cbfabb1a6f089eb9c585ef52147de01959c

closed 2024-06-11 by ruevs