solvespace/solvespace #862: Enhancement: partial bounded constraints

0x0203, 2020-12-17:
When making assemblies for mechanical devices, it might be helpful to still allow some degrees of freedom, but to put bounds on them to restrict the range of motion of the assembly. For example, putting a partial dimensional constraint on a line such that its length could be anywhere in the range [x,y] (or x, +/- y), or bounding the angle between two lines between x and y degrees. 

Feel free to close if this isn't feasible/practical given the structure/design philosophy of the code, or if the same can already be achieved in other ways.

kk6mrp, 2020-12-17:
Are you referring to this: https://github.com/solvespace/solvespace/issues/81?

0x0203, 2020-12-17:
Ah, many thanks. Looks like people are already on it.

closed 2020-12-17 by 0x0203

phkahler, 2020-12-17:
I wouldn't say they're on it. That was a long time ago and it's still not in. :-(

0x0203, 2020-12-17:
True enough. Until 3.0 gets the final stamp of approval though, I'm okay waiting. After that, hopefully it can be revisited.