solvespace/solvespace #863: Toggling reference property via property browser doesn't update error state

0x0203, 2020-12-17:
Fixing an over-constrained sketch by converting the offending constraint to a reference constraint via its checkbox in the property browser doesn't update the error state until something in the model is changed. The problem doesn't happen if the constraint is simply selected and the 'e' key is pressed to toggle reference. 

### System information
SolveSpace version: 3.0~cac3aae0 (openmp version)
Operating system: Windows 7

### Expected behavior
When an offending constraint that over-constrains the model is converted to a reference via the property browser, the state should update and remove the error displays. 

### Actual behavior
The system continues to show an error is present, directing you to a list of conflicting constraints and turning the edit window red.

### Additional information
This seems to be a problem both in moving to/from the error state when changing a constraint's reference property via the property browser. An easy way to demonstrate this is with triangle of sides constrained to vertical d3, horizontal d4, and hypotenuse d5. Toggling the d5 constraint reference property should show this off.

ruevs, 2020-12-18:
This is true. There are more than one situations like this. That is why there is "Edit | Regenerate All" (or press space). In the past there were even more situations like this some were slowly removed...

robnee, 2021-02-07:
I think I have a one-line fix for this.  I tried just `SS.ScheduleGenerateAll()` and that works for turning reference checkbox on but when toggling reference checkbox off the screen doesn't turn red unless the group is marked dirty to force the group to resolve.

```
diff --git a/src/describescreen.cpp b/src/describescreen.cpp
index 17f7361..03fafee 100644
--- a/src/describescreen.cpp
+++ b/src/describescreen.cpp
@@ -47,6 +47,7 @@ void TextWindow::ScreenConstraintToggleReference(int link, uint32_t v) {
     SS.UndoRemember();
     c->reference = !c->reference;

+    SS.MarkGroupDirty(c->group);
     SS.ScheduleShowTW();
 }
```

>The problem doesn't happen if the constraint is simply selected and the 'e' key is pressed to toggle reference.

Not always true.  There is a problem with `ScheduleGenerateAll` and `ScheduleShowTW` at least on Windows.  Both are scheduled by setting a zero millisecond timer.  These fire once all message processing stops, typically several tens of milliseconds later.  The problem is the timers are independant and don't fire in a predictable order and there is no guarantee that the first timer set will be the first one that fires.  This sometimes causes the text window to try to refresh **before** the geometry is regenerated.  I added some logging and I can see when GenerateAll is deferred the text window doesn't update properly.

@ruevs @phkahler I will submit a PR for the mark dirty fix but I am also working on a fix for the timer problem.  I combined the the two timers into a single refreshTimer.  `ScheduleGenerateAll` and `ScheduleShowTW` each set a separate refresh flag and sets the refreshTimer.  When it fires it checks the flags and ensures that `GenerateAll` happens first.  I does work but there may be a better solution.

phkahler, 2021-02-07:
@robnee  I'm going to merge the one-liner #926. Do you consider that sufficient to close this? It seemed to correct it for me, but your timer concern sounds non-deterministic so maybe not?

robnee, 2021-02-07:
Yes. the PR does close this issue.  The timer issue is isn't related to toggling with the property browser.  I just happened to noticed it while trying toggling with the 'E' key.

closed 2021-02-08 by phkahler