solvespace/solvespace #864: Exiting the segment and bspline tools leaves an additional line segment

0x0203, 2020-12-17:
### System information
SolveSpace version: 3.0~cac3aae0 (openmp version)
Operating system: Windows 7 (Also happens on Windows 10)

### Expected behavior
When using the line segment or bspline tool, to finish using the tool, right click or escape can be pressed to stop drawing. The last left-clicked point should be the final point of the segment or spline chain. 

### Actual behavior
An additional segment is left with the final point wherever the mouse is at, leaving an additional point/segment. I could sort of understand the final point must be made with right click if that were documented, but dropping the point wherever the mouse happens to lie when escape is pressed seems to be pretty unexpected behavior.

phkahler, 2020-12-18:
The right click behaviour is not a bug. Escape might be.

ruevs, 2020-12-18:
Bhis is behaviour by design. It has been like this since at least version 1.9

closed 2020-12-18 by ruevs