solvespace/solvespace #865: Rotating a "link / assembly" does not work with ctrl / shift down from the start

vespakoen, 2020-12-18:
### System information

SolveSpace version: edge

Operating system: macOS

### Expected behavior

When dragging a point, with ctrl or shift down, I would expect the model to rotate.

### Actual behavior

It doesn't, the only to get it to rotate is to start dragging without ctrl / shift down, then, while dragging add ctrl / shift to the combo and it starts working.

### Additional information

This is very confusing, It took me a while to figure out how to get it working consistently.

In `mouse.cpp`, there is a `if (!havePainted) { ... return; }`, when removing this, it works the way I expect it to work, but this is of course not the right solution.

I am not exactly sure what should be done to fix it.

ruevs, 2020-12-18:
Directly `Shift+LeftDrag` rotates the model around the selected point as expected for me (on Windows).

As you described `Ctrl+LeftDrag` only works if I am already `LeftDragging`. It works rather erratically, it can stop rotating the model "spoadically". And the rotation it does is the same as `Ctrl+WheelClickDrag" for the whole view. This is rather interesting - I did not even realize this function existed...

robnee, 2021-02-27:
I botched the first PR but but I've resubmitted a clean one.  There's a lot going on in the mouse movement handling code with many code paths.  Some code paths skip important steps like marking geometry dirty while dragging.  When that happens the havePainted check can "lock up" because SS gets into a state where it doesn't want to repaint (which would reset havePainted) but it also isn't marking anything dirty so no painting is deemed to be required.  This fix makes sure that the geometry gets consistently marked dirty (through all code paths) when dragging.

phkahler, 2021-03-02:
Does 945 (merged) close this?

vespakoen, 2021-03-02:
Yes it does 🎉, closing...

closed 2021-03-02 by vespakoen