solvespace/solvespace #866: Discuss approach to FEA in solvespace

phkahler, 2020-12-19:
Someone has recently been trying to display FEA results using SolveSpace by linking a mesh and modifying our rendering code. This invites the idea of doing more FEA related stuff in SolveSpace so I thought we could discuss that idea here.

As I see it, the renderer needs to be modified to interpolate an additional (heat, stress, etc...) parameter across triangles and use that value to index a color map as you can see in any search for [FEA results.](https://www.google.com/search?q=FEA+result&client=firefox-b-1-d&source=lnms&tbm=isch&sa=X&ved=2ahUKEwig9onN3trtAhUUK80KHV4aDk4Q_AUoAnoECAoQBA&biw=1175&bih=1008)
The other thing it needs to do is add a displacement to each vertex to show deformation. This displacement is often exaggerated (or even animated) to help visualize what is happening.

My thought on rendering is that we could add those features and leave them enabled full-time. The color map could be an RGBA map that we just make transparent in normal use. The displacement vectors would be zeroed and their multiplier would also be zero. The SolveSpace renderer seems fast enough that this would make no visible difference.  If displacements were done in a shader we probably couldn't even measure a difference, but that may bump our OpenGL version.

Beyond rendering is the question, how far should we go? Should FEA meshes be generated in SolveSpace? We could borrow code for 2d and 3d meshing from Gmsh which is also under GPL. Should we then add the ability to apply load conditions (or heat or whatever) in SolveSpace and just use an external FEA solver?

My main concern in heading this direction is maintaining the quality CAD experience SolveSpace is becoming known for, as well as keeping the code small and manageable with few resources. This would be a very large undertaking, but we might be able to go in smallish steps that each have some (limited) utility.

@ceanwang @jwesthues Thoughts?

jwesthues, 2020-12-19:
No strong feelings myself. I see the appeal of the feature, and e.g. of using the existing constraint/relation UI to apply boundary conditions. It's a big effort, though.

One possibility might be to add the UI for specifying conditions and viewing results, but keep the meshing and FEA solver itself as an external program, invoked from SolveSpace but appearing in a separate command prompt window? I'm thinking sort of analogous to invoking a compiler from within an IDE.

phkahler, 2020-12-20:
> One possibility might be to add the UI for specifying conditions and viewing results, but keep the meshing and FEA solver itself as an external program, invoked from SolveSpace but appearing in a [...]

I was thinking the UI is probably the hardest part, so I was looking for a reasonable progression to a goal of doing almost all of it with that being close to last ;-)  Is there a format to output geometry **and** conditions prior to meshing? Does STEP support FEA features like that?

There are reasons FEA has traditionally been done outside of CAD. What the UI should be depends on what type of simulation you want to do. Thermal, structural, CFD all seem to want unique UI features in addition to people wanting/needing to use different solvers. CFD even wants different visualizations so I think that's out of the question unless someday we support the others first.

ceanwang, 2020-12-22:
Yes, keep the meshing and FEA solver itself as an external program.

For meshing, I tried to use 'system' call, but need to pass on the file name to it. I want to export the model to a Gmsh's geo file. The file name is determined by the 'savefile' variable with a geo extension. Something like this:
`system("gmsh",savefile);`
But only system("gmsh") works.

phkahler, 2020-12-22:
@ceanwang  So you want to apply boundary conditions in SolveSpace, otherwise you'd just export a STEP file and do everything in Gmsh.  I asked above, is that even possible to export the geometry and conditions in some format? Which one? what all is in a gmsh savefile?

The other reason to create the mesh in SolveSpace is that we'd create the triangles directly in our existing mesh and then append the results data from the FEA results file. No linking - results right on the model as created in SolveSpace. We might also get a better triangle surface mesh by integrating the code from Gmsh. This also skips an entire tool in your workflow  SS->Solver->SS  rather than SS->Gmsh->Solver->SS.  If that's what you want I still don't see what's wrong with SS->Gmsh->Solver->Gmsh instead which can be done right now.

ceanwang, 2020-12-22:
Yes, I want to use SS to apply boundary condition and  export the geometry and conditions into Gmsh's 'geo' file format. First, just test to export geometry into a Gmsh's geo file.

SS may only has 2d mesh, FEA also need 3d mesh. It's OK to use SS to meshing, just the problem FEA could solve maybe limited. But if it is easy to implement, why not?

phkahler, 2020-12-23:
I'm not familiar with that geo format. If it can do NURBS and boundary conditions that'd be great. I guess just the triangle mesh and boundary conditions would be OK for that workflow too - that would indeed be a useful approach.

>> SS may only has 2d mesh, FEA also need 3d mesh. It's OK to use SS to meshing, just the problem FEA could solve maybe limited. But if it is easy to implement, why not?

SolveSpace 2D mesh is not suitable for FEA, however it is licensed as GPLv3 while Gmsh is GPLv2 or later, so we could rip both 2D and 3D mesh generators out of Gmsh and put them directly in SolveSpace if we wanted to. I don't know how easy that would be, but I think it might be easier than adding all the GUI stuff to apply conditions, but those will be needed either way.  BTW we will also need the ability to define material properties. There is a LOT of work to bring FEA in any form :-)

ceanwang, 2020-12-23:
I have no idea of what a SS triangle mesh looks like. Is there an example? If I draw a rectangle, could I mesh it in SS?

ceanwang, 2020-12-23:
https://github.com/ceanwang/solvespace/blob/89867daf9d892a5834d2145a038a5464c102e801/src/export.cpp#L1266

ilguido, 2025-05-20:
In my humble opinion Solvespace should focus on what it does best, that is modelling. I am a fan of the UNIX philosophy "make each program do one thing well". That said, a better integration in a FEM workflow would be greatly appreciated (at least by me), so, as a starting point, better support for file exports (that is actually good for any workflow). A Gmsh geo file is actually a script: it can be used for geometry and much more, so that would be a nice addition with much potential.