solvespace/solvespace #871: Error computing extrude-difference on step-rotated assembly

dakaufma, 2020-12-26:
### System information

SolveSpace version: 3.0~a2b5d0d4

Operating system: Arch Linux

### Expected behavior

g00a-extrude should cut a notch for each of its 6 profiles.

### Actual behavior

g00a-extrude cuts a notch for profiles that intersect exactly one copy of the step-rotated part, and fails to construct geometry for notches that bridge two copies of the step-rotated part. It outputs the error `failed to assemble polygon to trim nurbs surface in uv space` on the command line.

[image]

### Additional information

I also tried setting the g003 import to "union" instead of "assemble" but the union operation fails.

[slip_disk.zip](https://github.com/solvespace/solvespace/files/5743853/slip_disk.zip)
(main file is outside.slvs, which links profile_outside.slvs)

ruevs, 2020-12-27:
Note to self - analyse if this belongs to an existing type of failure and add it here https://github.com/solvespace/solvespace/issues/738 .

phkahler, 2020-12-28:
Looking closely, the rotate 9x group failed to merge coincident surfaces in the area where the later failures occur. Here is a view highlighting a surface that should not exist:
[image]
There are still 2 coincident surfaces in there.

ruevs, 2021-01-08:
In fact two surfaces co-exist there with opposite normals. This is a typical "singularity" case... Like the spoon here https://github.com/solvespace/solvespace/issues/470

ruevs, 2021-01-08:
https://github.com/solvespace/solvespace/issues/470#issuecomment-527175273

phkahler, 2021-01-09:
@ruevs it's not really the same as #470 since the problem actually happens in the Rotate group here. The opposite facing surfaces should have been eliminated in that group. I think most of those surfaces get eliminated properly, but not these two pairs for some reason.

I tried making a 22.5 degree wedge by revolving a rectangle (not around its own edge), then did a rotate group to make multiple copies and constrain so they are coincident. It failed very very badly.