solvespace/solvespace #872: No equal angle constraints larger than 180 degrees

0x0203, 2020-12-28:
SolveSpace version: 3.0 ~2939abf5
Operating system: MX Linux 19.3 (kernel 4.19.0)

I don't know if this is a bug or if it's operating as designed; I suspect the later. That said, it would be nice to be able to have equal angle constraints where the calculated angle can be greater than 180 degrees. While I understand that normally the complementary angle is equivalent as far as drawing is concerned,  I've run into a situation where it isn't sufficient, particularly when using equal angle constraints to control the movement of a model via a control point. The attached is a simplified example where the rig on the left is the model rig, and the lines on the right are the control rig, used to simplify moving the rig around through its range of motion. By moving the control rig, it's impossible to move the model joints through more than 180 degrees. 

Barring a direct fix for the angle constraints, is there an easier/better way of achieving what I'm trying to do? I.e.: moving a single joint in a chain of linkages where the angles between all down-stream links don't change, for example an excavator bucket or robotic arm.

[angle_demo.zip](https://github.com/solvespace/solvespace/files/5748967/angle_demo.zip)

ruevs, 2020-12-28:
I cant try your model right now, but maybe "Constrain | Other Supplementary Angle `U`" is what you are after? (while the equal angle constraint is selected)

0x0203, 2020-12-29:
I've tried playing with the supplementary angle, but it has the same issue. I think that ultimately, the way it's calculating the angle between two lines as you drag one segment's end point around will always result in an angle less than 180. Which is enough for drawing as you can always just use the supplementary angle to put things where they need to be, but in this case, I'm moving a line around and I'm wanting it to be able to represent everything between 0 and 360 degrees. Hopefully the model will help explain... I'm having a hard time putting it into words.

0x0203, 2020-12-29:
Now with a demo video to help explain:

https://user-images.githubusercontent.com/44345347/103288275-82a83480-49b2-11eb-9928-85f1e460cf7f.mp4

0x0203, 2020-12-29:
Now with demo video to help explain: 

https://user-images.githubusercontent.com/44345347/103288543-1417a680-49b3-11eb-8b29-c5874a3b316c.mp4

0x0203, 2020-12-29:
Now with demo video to hopefully help explain:

https://user-images.githubusercontent.com/44345347/103291878-b38c6780-49ba-11eb-8810-792c87565e3a.mp4

phkahler, 2020-12-29:
Speculation. What if you made a small  line perpendicular to the "reference" line in the lever and one in the bucket. Use two equal angle constraints - one as is, and one using the perpendicular lines. You'll have to check the "allow redundant constraints" box for this group. Maybe that will work?

0x0203, 2020-12-29:
Ah, that's a clever workaround. Adding the perpendicular lines does indeed seem to work. It's also fixing some of the redundant/unsolvable constraint errors that I was getting when I had angles of 0/180 without the perpendicular lines. Many thanks.

0x0203, 2020-12-29:
Although, after playing around with it a bit more, when constraining the lengths of the perpendicular lines it seems to be running into some numerical precision issues as there are many places where it complains about unsolvable constraints. Leaving the perpendicular lines unconstrained doesn't seem to cause any problems. Not a big deal; just a little less tidy than preferred.

phkahler, 2020-12-29:
This issue may fall under issue #76.

Another way to go would be 2 point-line-distance equality constraints (do we have those?) but then you need to ensure that the coordinates of a point (relative to two perpendicular lines) is always the same sign - never cross those reference lines. That would allow following a circle with no redundant constraints. You can trace a small circle on a long arm by constraining an extra point that follows the "level" to the line that follows the angle.

We should still have a better way to do this.