solvespace/solvespace #873: error while building edge release

inet56, 2021-01-05:
os: Arch Linux

running `cmake .. -DCMAKE_BUILD_TYPE=Release` produce error

<details>
<summary>output of <code>cmake .. -DCMAKE_BUILD_TYPE=Release</code> </summary>
<pre>
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_FILE_PREFIX_MAP
-- Performing Test HAS_FILE_PREFIX_MAP - Success
-- Using in-tree libdxfrw
-- Using in-tree mimalloc
-- Looking for backtrace
-- Looking for backtrace - found
-- backtrace facility detected in default set of libraries
-- Found Backtrace: /usr/include  
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/libpng.so (found version "1.6.37") 
-- Found Freetype: /usr/lib/libfreetype.so (found version "2.10.4") 
-- Checking for module 'cairo'
--   Found cairo, version 1.17.4
-- Found OpenGL: /usr/lib/libOpenGL.so   
-- Found SPACEWARE: /usr/lib/libspnav.so  
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.13.91
-- Checking for module 'json-c'
--   Found json-c, version 0.15
-- Checking for modules 'gtkmm-3.0>=3.18;pangomm-1.4;x11'
--   Found gtkmm-3.0, version 3.24.3
--   Found pangomm-1.4, version 2.42.2
--   Found x11, version 1.7.0
-- Configuring incomplete, errors occurred!
See also "/home/tnv/t/solvespace-edge/build/CMakeFiles/CMakeOutput.log".
</pre>
</details>

<details>
<summary><code>CMakeOutput.log</code></summary>
<pre>
The system is: Linux - 5.9.13-arch1-1 - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/tnv/t/solvespace-edge/build/CMakeFiles/3.19.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
[... 401 lines cut ...]
Run Build Command(s):/usr/bin/make cmTC_9e227/fast && /usr/bin/make  -f CMakeFiles/cmTC_9e227.dir/build.make CMakeFiles/cmTC_9e227.dir/build
make[1]: Entering directory '/home/tnv/t/solvespace-edge/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_9e227.dir/CheckSymbolExists.c.o
/usr/bin/cc   -ffile-prefix-map=/home/tnv/t/solvespace-edge=.  -o CMakeFiles/cmTC_9e227.dir/CheckSymbolExists.c.o -c /home/tnv/t/solvespace-edge/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_9e227
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_9e227.dir/link.txt --verbose=1
/usr/bin/cc  -ffile-prefix-map=/home/tnv/t/solvespace-edge=.  -Wl,--as-needed   CMakeFiles/cmTC_9e227.dir/CheckSymbolExists.c.o -o cmTC_9e227 
make[1]: Leaving directory '/home/tnv/t/solvespace-edge/build/CMakeFiles/CMakeTmp'

File /home/tnv/t/solvespace-edge/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <execinfo.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef backtrace
  return ((int*)(&backtrace))[argc];
#else
  (void)argc;
  return 0;
#endif
}
</pre>
</details>

Akaame, 2021-01-09:
Hello,

Could not reproduce this on a Ubuntu with more or less the same build env. Could you provide any additional files such as CMakeError.log. Your CMakeOutput.log is almost identical to a successful build output which does not provide value as far as diagnostics go.

inet56, 2021-01-12:
I couldn't find any file named CMakeError.log in build directory or any of its sub directory.

inet56, 2021-01-12:
It would be nice if there was also a `configure` script and a `Makefile`

ongun-kanat, 2021-01-12:
@inet56 CMake produces a lot more output if there is an error. Did you run

`cmake .. -DCMAKE_BUILD_TYPE=Release > some_out` rather than `cmake .. -DCMAKE_BUILD_TYPE=Release`. If you do so, please redirect stderr output too by appending `2>&1` `cmake .. -DCMAKE_BUILD_TYPE=Release > some_out 2>&1`

inet56, 2021-01-12:
content of some_out after running `cmake .. -DCMAKE_BUILD_TYPE=Release > some_out 2>&1`. 
<details>
<summary>some_out</summary>
<pre>
CMake Error at cmake/GetGitCommitHash.cmake:8 (file):
  file failed to open for reading (No such file or directory):

    /home/tnv/t/solvespace-edge/.git/HEAD
Call Stack (most recent call first):
  cmake/GetGitCommitHash.cmake:35 (get_git_commit_hash)
  CMakeLists.txt:39 (include)

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_FILE_PREFIX_MAP
-- Performing Test HAS_FILE_PREFIX_MAP - Success
-- Using in-tree libdxfrw
CMake Error at CMakeLists.txt:182 (add_subdirectory):
  The source directory

    /home/tnv/t/solvespace-edge/extlib/libdxfrw

  does not contain a CMakeLists.txt file.

-- Using in-tree mimalloc
CMake Error at CMakeLists.txt:189 (add_subdirectory):
  The source directory

    /home/tnv/t/solvespace-edge/extlib/mimalloc

  does not contain a CMakeLists.txt file.

-- Looking for backtrace
-- Looking for backtrace - found
-- backtrace facility detected in default set of libraries
-- Found Backtrace: /usr/include  
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/libpng.so (found version "1.6.37") 
-- Found Freetype: /usr/lib/libfreetype.so (found version "2.10.4") 
-- Checking for module 'cairo'
--   Found cairo, version 1.17.4
-- Found OpenGL: /usr/lib/libOpenGL.so   
CMake Warning (dev) at /usr/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args` (SPACEWARE)
  does not match the name of the calling package (SpaceWare).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindSpaceWare.cmake:19 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:280 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found SPACEWARE: /usr/lib/libspnav.so  
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.13.91
-- Checking for module 'json-c'
--   Found json-c, version 0.15
-- Checking for modules 'gtkmm-3.0>=3.18;pangomm-1.4;x11'
--   Found gtkmm-3.0, version 3.24.3
--   Found pangomm-1.4, version 2.42.2
--   Found x11, version 1.7.0
-- Configuring incomplete, errors occurred!
See also "/home/tnv/t/solvespace-edge/build/CMakeFiles/CMakeOutput.log".
</pre>
</details>

I am building from edge release.

ppd, 2021-01-12:
@inet56 you need to initialize the git submodules. The release tarball will not build without you manually adding those submodules. Better checkout the repo as per the instructions.

inet56, 2021-01-12:
Have built it following the instructions on README.

But there is no uninstall target in the generated Makefile.

ongun-kanat, 2021-01-12:
Cmake does not generate uninstall targets. Installing software to system directly is a dangerous move even a uninstall target does exist. That's why package managers like pacman, apt etc. are used for installing software to your system. If you want an easy uninstall option write a PKGBUILD file for pacman.

closed 2021-01-12 by inet56

inet56, 2021-01-12:
> Installing software to system directly is a dangerous move even a uninstall target does exist

It can be, but it is done by the root user, who can do anythings to the system.