solvespace/solvespace #874: No real RC1 tarball

vuokko, 2021-01-06:
### System information

SolveSpace version: 3.0RC1

Operating system: OpenBSD -current

### Expected behavior

Extracting source tar ball should be buildable

### Actual behavior

github is providing automatic tar balls. (snapshots of source) and extlibs aren't populated.
Trying to populate them with git ends up in:
git submodule update --init extlib/libdxfrw extlib/mimalloc
fatal: not a git repository (or any parent up to mount point /)

### Additional information

The instructions to build on OpenBSD are so simple and I decided to try write a port for solvespace and do some testing. 
So that's why requirement of source release.

ppd, 2021-01-06:
I suppose it's possible to publish another tarball asset including the submodules. Or we could provide a script that downloads the submodules without requiring the tarball to include a git repository.

ruevs, 2021-01-16:
https://github.com/solvespace/solvespace/issues/772#issuecomment-730461961

phkahler, 2021-05-16:
Closing as a duplicate of #1028 which is not specific to RC1.

closed 2021-05-16 by phkahler