solvespace/solvespace #875: Crash on Arc-Line tangent constraint

johreh, 2021-01-07:
### System information
SolveSpace version: 3.0~14e837a
Operating system: Arch Linux

### Steps to reproduce
1. Create a line and an arc that share a vertex
2. Select arc first
3. Select line second
4. Create a tangent constraint

### Actual behavior
Solvespace crashes

### Additional information
This is due to the "swap" call in constraint.cpp being moved in commit 14e837a. The swap has become local to Constraint::ConstrainArcLineTangent which causes "entityA" and "entityB" to be assigned the wrong way around when execution returns to Constraint::MenuConstrain.

ruevs, 2021-01-08:
My stupidity...
Reverted my commit here: https://github.com/solvespace/solvespace/pull/877 3e3ccdca8d6c66080078c46c274cbde645f43cf4

closed 2021-01-08 by ruevs

phkahler, 2021-01-08:
@ruevs  thanks for the quick fix!

ruevs, 2021-01-09:
Sorry for the stupid code :-)