solvespace/solvespace #876: X Window error on Raspian

someSven, 2021-01-08:
### System information

SolveSpace version: Raspian repo, apt-cache show solvespace gives me "Source: solvespace (2.3+repack1-3), Version: 2.3+repack1-3+b1"
Operating system: Raspian on a Rapberry Pi 3

### Expected behavior
Solvespace shouldn't crash every few minutes, when doing some operation, like creating a new workplane or adding or deleting a constraint.

### Actual behavior
This is the error message in the shell:
The program 'solvespace' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 4511 error_code 8 request_code 130 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

### Additional information
I also tried gdb (Gnu Debugger), but it was the first time in my life. Only with the <run> command, nothing more.
Output: Bunch of threads exited, and a this line again and again: "warning: JITed object file architecture armv8-a is not compatible with target architecture armv6." I might ask in the Raspian forum as well if that helps, maybe they compiled or repacked it wrong. I used a Raspberry Pi 3, it works sufficiently well otherwise, at least for the small parts I tried it with. It also might matter that I'm running it with OpenGL (I think with full KMS, but I'm not sure).

[PM35S_cased_lever_01.stl.zip](https://github.com/solvespace/solvespace/files/5784673/PM35S_cased_lever_01.stl.zip)

ruevs, 2021-01-08:
I recently played with SolveSpace on a RaspberryPi 3 by building the latest master from source. It worked mostly fine.
https://github.com/solvespace/solvespace/pull/825#issuecomment-734509284
https://github.com/solvespace/solvespace/pull/825#issuecomment-734899712

By now 2.3 is rather old, would you be willing to [build from source as described in the readme](https://github.com/solvespace/solvespace#building-for-linux) and try it on your version of Raspbian?

someSven, 2021-01-08:
I'm looking into building it. This would not solve the problem for other users, though.

phkahler, 2021-01-08:
> I'm looking into building it. This would not solve the problem for other users, though.

We will not be making updates to version 2.3. If there are issues with 3.0 RC1 we would like to know. I want 3.0 done by mid March at the latest. Sooner is better.

ruevs, 2021-01-08:
I do not know when the Raspbian maintainers update packages but it would seem logical, that they do it when there is an official release. If that is the case until 3.0 is released building from source or getting it from the Snap store (https://snapcraft.io/solvespace) may be the only options. 

http://archive.raspbian.org/raspbian/pool/main/s/solvespace/

closed 2021-01-08 by ruevs

rpavlik, 2021-01-18:
I assume Raspbian Buster may not receive it, though I might be able to ask for it to be added to backports in Debian which I think gets enabled or at least it's compatible? Our goal is to get something approximately equal to 3.0 in Debian Bullseye which will eventually form the basis of the next version of Raspberry Pi OS, I would assume.

phkahler, 2021-01-18:
> Our goal is to get something approximately equal to 3.0 in Debian Bullseye which will eventually form the basis of the next version of Raspberry Pi OS, I would assume.

RC2 is out, and I consider that complete for 3.0. I'm just hoping to get updated translations #890 before making v3.0 final.  Optionally we might get some icon updates, but I'm not waiting for those.

We probably should make a 3.0 branch for backporting minor things, and if we can get into a major Debian release by creating that now I'd be OK with that. We can't even be certain all the translations will get updated.