solvespace/solvespace #879: Tool tips don't show after moving window

phkahler, 2021-01-09:
### System information
SolveSpace version: 3.0~d45e2c4c
Operating system: Fedora 33 (Linux) Wayland

### Expected behavior
Tool tips on icons should always show when hovered over.

### Actual behavior
After dragging the main window, the toolbar icons don't show their tips.
After dragging the text window those toobar icons don't show their tips.
After dragging either window, tools in the other window will show their tips (sometimes).

Bottom line, some combinations of clicking and dragging will enable or disable tool tips.

### Additional information

robnee, 2021-01-24:
There is a compiler warning that's related to a call to SetTooltip.  Perhaps this is an uninitialized variable problem?  I can't readily test this on linux but thought I would point this out:

/home/rnee/solvespace/solvespace/src/textwin.cpp: In member function ‘void SolveSpace::TextWindow::DrawOrHitTestIcons(SolveSpace::UiCanvas*, SolveSpace::TextWindow::DrawOrHitHow, double, double)’:
/home/rnee/solvespace/solvespace/src/textwin.cpp:648:31: warning: ‘hoveredY’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  648 |             window->SetTooltip(hoveredButton->Tooltip(), hoveredX, hoveredY, 28, 28);
      |             ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rnee/solvespace/solvespace/src/textwin.cpp:648:31: warning: ‘hoveredX’ may be used uninitialized in this function [-Wmaybe-uninitialized]

phkahler, 2021-01-25:
@robnee Thanks for looking, but it doesn't behave like uninitialized stuff. The missing tips will come back with some focus changes and window moving. They will then stop again with more similar manipulation.

robnee, 2021-02-07:
@phkahler does the hover highlight around the icon still appear even when the tool tips are missing?

robnee, 2021-02-07:
This patch adds logging to show what's going on here.  The tracking of enter and leave events is flaky and ends up clearing `_is_under_cursor` incorrectly sometimes.  I'm not sure it this tracking is necessary.  It might be related to #406 according to blame.

```
[rnee@centos8 build]$ git diff
diff --git a/src/platform/guigtk.cpp b/src/platform/guigtk.cpp
index 1f0cc55..7e0a1ac 100644
--- a/src/platform/guigtk.cpp
+++ b/src/platform/guigtk.cpp
@@ -805,12 +805,14 @@ protected:
     }

     bool on_enter_notify_event(GdkEventCrossing* gdk_event) override {
+        printf("Enter: %x  %.0f %.0f\n", gdk_event->window, gdk_event->x, gdk_event->y);
         _is_under_cursor = true;

         return true;
     }

     bool on_leave_notify_event(GdkEventCrossing* gdk_event) override {
+        printf("Leave: %x  %.0f %.0f\n", gdk_event->window, gdk_event->x, gdk_event->y);
         _is_under_cursor = false;

         return true;
```

phkahler, 2021-02-10:
Icon highlights still work when the tooltip do not.

robnee, 2021-02-11:
The highlight is done directly by the hittest routines in toolbar and textwin which sets the tooltip text and position and the same time.  It's the `is_under_cursor` tracking that ends up suppressing the display of the tooltip.  It's listening to enter/leave events 

If I change `on_leave_notify_event` to set `_is_under_cursor = true;` which disables the tracking and prevents the tracking bug then the problem appears to go away.

However, this was added to fix #406 which @rpavlik submitted.  I hesitate to touch this because I don't really understand the original issue but wanted to share what I found.

rpavlik, 2021-02-11:
Oh man that was a long time ago. I'm inclined to think it's something about using it on Wayland, since I don't recall seeing that on X. Or it might be a relative of my impossible to track down "property window doesn't stay on top but let events thru to the main window" bug that's pretty system specific.

robnee, 2021-02-11:
It's possible this work-around is no longer necessary but I'm afraid of removing this only to find that it is important to someone.  It seems to try to suppress sporadic tool tip popups. The problem now seems to be unreliable window enter/leave events.  The tool tip routine loses track of whether you're in or out of the window and suppresses some tool tips.  The reliability of the enter/leave events seems like it could be window manager or platform specific too, ugh.  There are a bunch of flags and things in the GTK CrossingEvent message that might be useful in making it more reliable so I'll look at that I think.  Thanks.

wolfv6, 2024-06-09:
Adding clarity to the tooltip behavior.

On SolveSpace left-column toolbar:
- When focus is on window, tool-tips do not display.
- When focus is on a different window, tool-tips display as expected.

On Property Browser toolbar:
- Tooltips sometimes show the same behavior and sometimes not.

Dragging a window, or not, makes no difference.
I am using SolveSpace 3.1 on Fedora 40, running Wayland.