solvespace/solvespace #880: Force distance constraints to plane while dragging

phkahler, 2021-01-09:
## Suggestion

When dragging a distance constraint, use SHIFT to force the constraint to the **nearest** workplane or face that contains the 2 points (or a point and line).  Another option might be to allow the user to select a face or workplane prior to dragging a constraint and have it stay within that plane.

### Use case
For technical drawing it looks good to have these items in-plane rather than at arbitrary angles to the sketch.

phkahler, 2021-01-10:
@Symbian9   I thought of this while looking at your usecase for "Dim Solid for Sketch Groups", so I'd like your thoughts in particular. I tried dimensioning a 3D sketch all in one group, and I kept needing to switch views to get the constraints to lie in the correct plane. Locking to the nearest axis-aligned plane might be enough? It is also interesting that radius/diameter constraints are already locked to the plane of an arc or circle.

Symbian9, 2021-01-11:
>  Another option might be to allow the user to
>
> I thought of this while looking at your usecase for "Dim Solid for Sketch Groups", so I'd like your thoughts in particular.

@phkahler, Yeah! I has planned to create such feature request few days ago.

JFTR, Actually in 3D mode sketching diameter constraint dimension is locked in circle/arc plane.

As for me, there should be few options:

1. Snap dimension constraint to plane in 3D (align distance constraint datum point to selected plane); 
2. Select a face or workplane prior to dragging a constraint and have it stay snapped within that plane;
3. `TBA` *(more suggestion soon!)*

### TBA

***NOTE:** I would add more details, as this feature is really important, but just in few days!* ;-)