solvespace/solvespace #886: Edge builds are stacking up

phkahler, 2021-01-10:
There should only be one edge build at any given time. If possible it would also be nice not to have it at the top of the list.  3.0RC1 would be the ideal top build?

ruevs, 2021-01-11:
Most of the time they do not stack up. But there is something that causes some of them to be kept sometimes. I have no idea what.

For a while it used to be only "Suggest sensible defaults in file dialogs" from Dec 10 (which I ignored, because it was in the early days of GitHub actions), but now I see that "Revert "Clean up paste transformed and constrain operations a bit." from Jan 8 has stuck around. Both of them are marked as "`Draft`" while the normal one is always "`Pre-release`".

For now I think we should just delete those two... but maybe @ppd could shed some light why it happens?

ppd, 2021-01-12:
Hm. Strange. The "Draft" releases are at least not publicly visible. There could be an issue with `actions/create-release@v1`; it should never create draft releases (because of `draft: false`).

ppd, 2021-01-12:
Were the now `Draft` releases ever in `Pre-release` state? If that were the case, it would point to a problem (possibly a race condition) in `dev-drprasad/delete-tag-and-release` where it should delete both the tag _and_ the release.

phkahler, 2021-01-12:
My best guess is that it's related to merging PRs too close together in time.  I recently merged at least two within a few minutes which I think are some of the lingering builds.

ppd, 2021-01-12:
Consider `CD #15`, which builds from 3e3ccdc and releases successfully to edge.
`CD #16` is "Add toolbar icons for Revolve and Helix created by jkrei0 in issue #857", and `CD #17` is "
Update ISSUE_TEMPLATE.md (#859) Improve formatting".

`CD #16` starts at 2021-01-09T17:27:30Z
`CD #17` starts at 2021-01-09T17:41:25Z

So they are indeed close. The `Update Edge Release` stages try to clean up the old release, which fails for `CD #16` at `Sat, 09 Jan 2021 17:49:03 GMT`. Creating the new release succeeds and everything else completes at `Sat, 09 Jan 2021 17:49:35 GMT`.

Assuming no clock difference between `CD #16` and `CD #17`, `CD #17`'s `Update Edge Release`  job runs at `Sat, 09 Jan 2021 18:00:56 GMT`, which is around ten minutes later. So there's no overlap or anything like that; the error is introduced by `CD #16` deleting the `edge` tag without deleting the corresponding release.

Why could that be? https://github.com/dev-drprasad/delete-tag-and-release/blob/6bf166de162e06a25bd2055f3be7b2f33f8b5cc8/index.js#L72 tells us that the action only considers non-draft release for deleting. 

Let's assume the GH API returns correct data, and the edge release is really in draft mode at that time. If that's true, something must have changed the release to `draft: true`. How would that be possible?

1. Manually in GH. Unlikely.
2. By overwriting the existing `edge` tag before removing the release. Did anyone force push their local tags to the remote? That would indeed orphan the release and relegate it to draft mode.
3. `actions/create-release@v1` created a draft release in the first place, for whatever reason.

phkahler, 2021-01-12:
Somewhere in there I pushed one of my branches to "upstream" instead of "origin" and I then deleted it and went the normal route. But that was not to master, it was a new branch.

ruevs, 2021-01-12:
> Were the now `Draft` releases ever in `Pre-release` state? 

I do not know. Have not paid attention.

I have force pushed changes to the remote (quickly one after the other) many times, but (I think) only to my fork. If I or anyone else (I think vespakoen did at one time) have by mistake pushed to solvespace this may explain it.

ppd, 2021-01-12:
One would have to push the tags explicitly anyway; they don't get transferred by the default `git push`. Let's delete the draft releases and see if it happens again.

ppd, 2021-02-04:
And it happened again. There's some consistency problem with the GH API seemingly returning no valid edge release to delete, I think. Will take another look in a little while.

ruevs, 2021-02-04:
I know what I did in this one:
https://github.com/solvespace/solvespace/releases/tag/edge
I added a new commit to an already opened pull request.

I'm not rite if the initial CI run from opening the pull request had finished when I added the second commit.

ruevs, 2021-03-06:
It still happens.

And now `Deploy ARM64 Snap` also fails
https://github.com/solvespace/solvespace/runs/2009558378?check_suite_focus=true
The last one that succeeded seems to have been this one https://github.com/solvespace/solvespace/actions/runs/546319493 from 27 days ago.

https://github.com/solvespace/solvespace/actions

ruevs, 2021-04-07:
Still pining up as "Draft" and when I view [the releases](https://github.com/solvespace/solvespace/releases) when not logged in there is no "Edge" release any more.

I'll do a "Edit | Publish release" on the latest one.

ppd, 2021-04-07:
Yes, the underlying issue remains. It seems the release action is unmaintained, and alternatives are recommended in their readme. I'll try an updated workflow in my fork to see whether the problem persists.

ppd, 2021-04-10:
I hope #1008 fixes this. I couldn't reproduce the issue in my fork with the included workaround.

ppd, 2021-12-21:
The edge releases are gone, and so this issue is not relevant any more (+ it was already solved by #1008).

closed 2021-12-21 by ppd

Symbian9, 2021-12-21:
> The edge releases are gone

Guess the last change should be related to this issue is to repalce "edge"/"cutting edge" with "nightly" in `README.md`:

- https://github.com/solvespace/solvespace/blob/2450010bbfa20a1ffbf5fd90b85c4b941f56cb16/README.md?plain=1#L69

ruevs, 2021-12-21:
> > The edge releases are gone
> 
> Guess the last change related to this issue which should be fixed is to replace "edge"/"cutting edge" with "nightly" in `README.md`:
> 
>     * https://github.com/solvespace/solvespace/blob/2450010bbfa20a1ffbf5fd90b85c4b941f56cb16/README.md?plain=1#L69

No. "Cutting edge" is an [idiom](https://www.oysterenglish.com/idiom-on-the-cutting-edge.html) [in English](https://www.theidioms.com/cutting-edge/) and the usage is correct.

Symbian9, 2021-12-21:
> No. "Cutting edge" is an idiom in English and the usage is correct.

I has never seen "cutting edge" release naming used by other popular FLOSS projects for its "nightly" builds :-/

FreeCAD uses "daily" term, but "nightly" is common for software releases.

Also, on linked page with idioms in English in "Example Sentences" section there NO any example of "cutting edge release" name use for something related to application releases and/or packaging ("technology" is more abstract than exact app nightly build).

ppd, 2021-12-22:
Also, nightly has a temporal meaning, i.e. nightly commonly refers to scheduled builds. We're building on each push to master.

phkahler, 2021-12-22:
"Leading Edge" might also be an interpretation of "Edge"

ruevs, 2021-12-22:
"Bleeding edge" then.